qmk/qmk_firmware #1686: Documentation: Write a guide to handwiring

skullydazed, 2017-09-06:
There are lots of guides that work for the hardware part, but setting up a hand-wired keyboard in qmk is easy if you know what you're doing, but we don't currently document that aspect very well for people who don't.

4sStylZ, 2017-10-12:
If you want some inspiration : https://imgur.com/a/qcgdF

fredizzimo, 2017-10-12:
That imagur link is actually a copy of [this geekhack.org thread](https://geekhack.org/index.php?topic=87689.0)

Gouty, 2019-10-01:
I'll take a crack at this

Gouty, 2019-10-01:
I made a good start to this, and I'm trying to combine the best bits of the community guides that are available.

Unless anyone says otherwise I'm going to hide all the in depth description about how a keyboard matrix works in `<details>` tags as it's not strictly needed to why it's done that way, just how.

Is there any way I can add images to this guide? I notice that images are listed in the .gitignore file, can I modify it or will it just be immediately rejected (and will it play nice with qmk.fm?)

drashna, 2019-10-02:
For images, hosting them on imgur or the like works.

Vexatos, 2019-10-02:
I'd suggest hosting them on github along with everything else. Having images specifically be hosted elsewhere would add dependency on another external website with little control over the file, and it would make a lot more sense for everything to be in one place; it would also subject the images to version control like everything else, meaning they too can be tracked.

jackhumbert, 2019-10-02:
We used to have images in the docs and in keyboard folders, but they significantly contributed to the repo size, which was painful for people just downloading the source code. [A semi-permanent hosting service like Imgur is our official recommendation for keyboard-related ones](https://docs.qmk.fm/#/hardware_keyboard_guidelines?id=imagehardware-files), but files can also be added to qmk.fm in a tandem PR. I think the latter would be appropriate for something like this, or any image needed in the docs. It may also be appropriate to put the guide itself on qmk.fm, or qmk_hardware (which could probably be fleshed-out a bit).

Vexatos, 2019-10-02:
Might it make sense to have a separate github repo for images?

noroadsleft, 2019-10-04:
I actually have [a repo I started](https://github.com/noroadsleft/qmk_images) because I had this very same issue of needing to include an image on something in qmk_firmware, but not wanting to actually put the image *in* qmk_firmware. (I agree strongly with the guideline of not allowing images in qmk_firmware.) Imgur works, but as has been said, it's an external dependency, and also doesn't version track.

I work with my repo mirroring the folder structure of qmk_firmware just because it makes sense to me to do it that way, which lowers the cognitive load of trying to figure out where an image is being used.

I wouldn't mind transferring ownership of that repo to QMK if it's something we decide would be good to do.

rishav142k, 2019-10-07:
I think it makes sense to have a completely separate repo just for hosting the images. That way, people downloading the source code won't have to deal with them.

Gouty, 2019-10-07:
By that logic, doesn't it make sense to have all the ancillary stuff (such as this documentation) as part of that separate repo? Or a wiki?

Gouty, 2019-10-07:
> @Gouty <https://github.com/Gouty> no, because then it would probably fall
> out of sync. When you check out the repo, you get the latest documentation.

Surely people are checking the website for the documentation (the content
for which is automatically updated from this repo)

yanfali, 2019-10-07:
@Gouty I think you just answered your own question. It's generated from a single source of truth and it comes bundled with the project. Having two projects would remove this property.

helios1101, 2019-10-07:
As I was explorinng the docs, I personally felt that everything has been bundled up at a single place, as images and important docs should be particularly in different places. Or either it may be, docs having a transverse relation to images must be at a separate place.

noroadsleft, 2019-10-07:
One reason that I'm personally against splitting the docs into a separate repo is that changes to core functionality (new features, extending/modifying existing features) often requires changes to the docs. Having them in the same repo means this can be done all at once in a single PR (excluding any need for images, obviously).

ThiagoWhispher, 2019-10-13:
Among the options discussed here, I would choose to keep in the same repository and make use of the wiki.

jasonappah, 2019-10-15:
Regarding the issue of images making the repo larger, would [Git LFS](https://www.atlassian.com/git/tutorials/git-lfs) solve the issue? That way, the images are still being versioned and can be edited, but they aren't being stored on GitHub. The tutorial I linked explains it way better than I have.

vaibhavbisht06, 2019-10-22:
Is there any way I can add images to this guide

4sStylZ, 2019-10-23:
Yes you can upload some image on imgur and use the markdown `![picture-name](url)` syntaxe to show them or, like somebody that I can remember upload them on another git repo and relate to them with the same syntaxe.

I suggest external link because it's not recommanded to upload image on the repo because this will increase the size of it to much.

sakshamtaneja21, 2019-10-26:
ill help in this

DavidYangDev, 2019-10-27:
We should keep the guide in the same repo since splitting up requires relinking and reconnecting everything, less work

skullydazed, 2020-01-25:
Addressed by #7044.

closed 2020-01-25 by skullydazed