qmk/qmk_firmware #1689: RGB Underglow - minimum brightness for breathing animation

drashna, 2017-09-07:
As title.  It would be nice to be able to set a minimum value (and possibly max) for the breathing animation so that it doesn't go completely dark. 

And handle it like the  RGBLED_BREATHING_INTERVALS, and similar values, so they could be defined in your keymap.

skullydazed, 2017-09-07:
This seems like a good idea to me. I took a stab at it with #1692, how's that look to you?

drashna, 2017-09-07:
Well, it looks like #1692 still allows it to go dark, but the overall animation for it looks better!

skullydazed, 2017-09-07:
Give `#define RGBLIGHT_EFFECT_BREATHE_CENTER 1` a shot.

skullydazed, 2017-09-07:
For reference, here is how `RGBLIGHT_EFFECT_BREATHE_CENTER` affects the breathing. The first image is the default setting of 1.85, which as you can see puts the lowest brightness at about 30/255, or ~12% (which may be below the threshold of visibility for your ws2812's.)

[image]

This second graph demonstrates how it looks when `RGBLIGHT_EFFECT_BREATHE_CENTER` is set to 1. As you can see the entire curve moves up, so the lowest brightness is 70/255, or ~27%, which should be visible on almost any board.

[image]

If you have OS X you can open 
[led_breathing.gcx.zip](https://github.com/qmk/qmk_firmware/files/1285333/led_breathing.gcx.zip) in grapher to play with the equation and get the curve to look how you want. You can from the images where to change `RGBLIGHT_EFFECT_BREATHE_CENTER`, but if you want to change `RGBLIGHT_EFFECT_BREATHE_MAX` too the correct number is the 255 on the right. We always need `x/255` to get the curve to fit our available 256 brightness levels.

drashna, 2017-09-07:
Perfect! Thanks!

closed 2017-09-07 by drashna