qmk/qmk_firmware #1694: faq_debug seems either out dated or not apprecable to ergodox or not sufficient

hiro2016, 2017-09-09:
https://docs.qmk.fm/faq_debug.html  
  
I followed the instruction and added `CONSOLE_ENABLE=yes` to `keyboards/ergodox/Makefile` and `qmk_firmware/Makefile`.    
It had no effect and the debug guide lead me to search for:  

 -  `magic key` - not much info on this(no links in faq_debug) except for   
https://github.com/tmk/tmk_keyboard#magic-commands and the really useful: https://docs.qmk.fm/getting_started_make_guide.html

- add `set debug_enable=true usually in matrix_init() in matrix.c.` in files  
  
- unplug usb cables  

After a long search and doubting, `I realized ergodox/rules.mk` had `CONSOLE_ENABLE   = no  # Console for debug(+400)` which was overriding what was set in Makefile.    
  
Thought of editing the doc and making a PR but posted just in case I'm still mistaken.

seebs, 2017-11-22:
This should be implicitly fixed now, because recent reorganizations mean there's no longer an ergodox/Makefile. All the rule stuff lives in rules.mk, so there's only one place to set it. FAQ is also now updated and referring to rules.mk.

hiro2016, 2017-11-23:
Thanks seebs,  I had forgotten to close this.

closed 2017-11-23 by hiro2016