qmk/qmk_firmware #1695: NKRO doesn't work with ergodox ez

lawkai, 2017-09-09:
I can only get 6kro by default but can enable NKRO by using left-shit-right-shift-n.
The NKRO_ENABLE is "yes" by default in the makefile.

Is there anything else I have to do to make NKRO enabled by default on ergodox-ez?

Note I have already tried to hold "KC_SPACE and KC_N" when plugging in my USB cable as mentioned in the FAQ but still can't get this to work.

hiro2016, 2017-09-10:
Searching the file contents under ergodox dir produces the following results and according to one doc, rules.mk takes precedence over Makefile configurations.   Maybe you have it set to no in one of rules.mk files? `rules.mk` file in your keymap folder should have the highest precedence, so try making one.

    ag "NKRO_ENABLE"
    keymaps/alphadox/Makefile
    6:NKRO_ENABLE      = yes # Nkey Rollover - if this doesn't work, see here: https://github.com/tmk/tmk_keyboard/wiki/FAQ # nkro-doesnt-work

    keymaps/videck/Makefile
    10:NKRO_ENABLE:=no

    keymaps/pvinis/Makefile
    4:NKRO_ENABLE = yes      # disable for windows

    keymaps/333fred/Makefile
    5:NKRO_ENABLE = yes

    rules.mk
    26:NKRO_ENABLE      = yes # USB Nkey Rollover - if this doesn't work, see here: https://github.com/tmk/tmk_keyboard/wiki/FAQ#nkro-doesnt-work
  
Yesterday I got caught over this too and wasted like a half day.      
DIY keyboard projects have been a tech savvy people only hobby and I guess docs are designed for them;but even then, I feel some cleaning might be needed. At least, users-guide and for-developers materials need some separation.

jackhumbert, 2017-09-10:
@lawkai you should be able to put `#define FORCE_NKRO` in your config.h to force it to be on all the time - it's disabled by default because some bioses aren't compatible with NKRO.

Unfortunately that FAQ is one area that has not been fully updated to how QMK works yet.

lawkai, 2017-09-10:
@hiro2016 Thanks, I had looked at the [rules.mk](https://github.com/qmk/qmk_firmware/blob/master/keyboards/ergodox_ez/rules.mk#L79) file in the  repo and it does indeed have NKRO_ENABLE = yes for ergodox-ez. However, it is still off by default.

@jackhumbert I can confirm that the `#define FORCE_NKRO` in config.h did the trick. Thanks!

closed 2017-09-10 by lawkai