qmk/qmk_firmware #1707: Shifting Tap Dance keys

mcroof, 2017-09-11:
Hey,
first of all thanks for the great program you guys put together and maintain.
I'm new to qmk and i tried the tap dancing feature. My plan was to make a single tap on `a` return an `a` while a double tap returns an `ä` (umlaut). 

Definition:
`[TD_A]  = ACTION_TAP_DANCE_DOUBLE(KC_A, KC_QUOT)` (KC_QUOT is ä on the german layout)
and then call the function in the keymap.

This does work very well but problems arise if I want to write capital letters. Single pressing `Shift + a` feels very sluggish and the `Shift` often does not get registered. Same for double tapping.
I compared the timing of my keypresses with "normal" keys which do get shifted correctly with a similar timing.
Is there any workaround for this or should I use a different function like macros or leader keys?

NateEag, 2017-09-11:
I've been ignoring this problem for months in my config, and just came here to finally file an issue for it.

Lo and behold, someone else had just brought it up.

Here is a bit more detail.

If an entire tap-dance sequence is completed within a certain span of time (I suspect related to `TAPPING_TERM`), then the unshifted keycodes will be sent, regardless of whether you held it down.

However, if you continue to hold down Shift after the tap dance, once the timeout is reached, then the Shifted keycode will be sent.

I don't have any insight into what causes this or what might fix it - just those additional reproduction details.

mcroof, 2017-09-12:
Very well explained.
I did some further testing with a longer `TAPPING_TERM = 550` (`kc` is unshifted `kc`, `KC` is shifted `kc`, `kcd` is tapped keycode, `KCD` is shifted tapped keycode): 

- shifting while tapping: 
`DOWN(kc), UP(kc), DOWN(Shift), DOWN(kc), UP(kc), 550ms, UP(Shift)` 
sometimes  `kc`, sometimes keeps `kc` pressed, sometimes `kc KC`

- shifting while tapping 2: 
`DOWN(kc), UP(kc), DOWN(Shift), DOWN(kc), UP(kc), UP(Shift), 550ms` 
mostly keeps `kc` pressed, sometimes `kc KC`

- shifting and canceling with other key 
`DOWN(Shift),  DOWN(kc), UP(kc), DOWN(kc), UP(kc), UP(Shift), DOWN(kc_other), UP(kc_other)` returns `kcd kc_other` (to be expected as the cancel of the tapping only happens with the `DOWN(kc_other)` so it doesnt register Shift

- shifting, canceling with other key 
`DOWN(Shift),  DOWN(kc), UP(kc), DOWN(kc), UP(kc), UP(Shift), DOWN(kc_other), UP(kc_other)` returns `KCD KC_OTHER`

NateEag, 2017-10-06:
Note that this bug may have gone unnoticed at first because the tap-dance feature's author, @algernon, uses Shift as a prefix key (tap to enable then press key to Shift):

https://github.com/algernon/ergodox-layout

Given that, it seems possible the combination of tap dance and shift wasn't exercised much when the code was written.

I could be entirely wrong - that's pure speculation.

I also wonder if this relates to #708, which appears to have been closed unsolved.

@algernon, do you have any insight or thoughts on this off the top of your head?

algernon, 2017-10-06:
Off the top of my head, to find the root cause, we need to look at when and how tap-dance events fire. In case of `Shift+TD`, the event will fire whenever the tap-dance timeout expires or is interrupted, by which time you likely released shift too. This is why you get a lower-case symbol when typing normally, but get an upper-case one if you hold shift through the tap-dance timeout.

The workaround here would be to create a tap-dance helper that checks the shift state when the tap-dance sequence begins, and if shift is pressed, it will make sure the same shift is also pressed when the tap-dance event fires (and then unregisters it, unless it is still held).

If anyone feels like they're up to the task, I'll happily give hints, and offer guidance, but I won't have time to do it myself in the near future.

NateEag, 2017-10-06:
I don't know if I'll have time to look at this any time soon, but it is an annoyance for me, so I may give it a shot at some point.

Thanks for the info!

pmalecka, 2017-12-01:
I took a stab at this here: https://github.com/pmalecka/qmk_firmware/commit/7f734bea0e8bf1e2eddb96514134b9367a0fbf65

@algernon would you have a look, if I missed something?

apturner, 2017-12-29:
I'm not sure if this is a manifestation of the same issue, but it seems that with `ACTION_TAP_DANCE_DOUBLE(KC_LSFT, KC_CAPS)`, Caps Lock will only enable if you double tap and hold until the `TAPPING_TERM` ends, whereas I would expect a quick double tap to enable it.

coreyzev, 2018-01-09:
@pmalecka if it works for you, you should just turn it into a pull request (if you havent already) 

https://github.com/qmk/qmk_firmware/blob/master/docs/contributing.md

After you do that one of the admins on this project will check it and give feedback

algernon, 2018-01-10:
@pmalecka looks good! Can you turn it into a pull request?

pmalecka, 2018-01-12:
thanks, I created a pull req: #2258

NateEag, 2018-01-20:
@pmalecka - I just rebased my personal keymap on top of master, and this is working for me now. Thanks so much! This has bothered me for ages.

mcroof, 2018-01-21:
@pmalecka Thanks a lot for implementing that! Works really well.

I'll close this issue as I consider this as solved.

closed 2018-01-21 by mcroof

pmalecka, 2018-01-22:
@NateEag @mcroof I appreciate the shoutout, glad it is working well for you guys.

NateEag, 2018-01-23:
@pmalecka I checked your profile but didn't find contact info, so throwing this out here - if you have a way to accept donations online, I'd like to send a few bucks your way as a more tangible thank you. You can reach me via nate@nateeag.com if you like. If you're not interested, no worries.