qmk/qmk_firmware #1709: LT variant suporting premature LT key release and LT down tap release cycle.

hiro2016, 2017-09-12:
I liked LT but it was too slow for me, so I wrote what I needed.    
  
I call it DLT.   
How it works:    
`DLT down, key tap, DLT up` -> as normal LT  
`DLT down, another key down,DLT up` -> DLT grabs keydown event and releases is as tap before DLT release signal is reported.      
`DLT down, another key tapped, yet another key tapped, DLT released` -> LT    
`DLT down, another key down, DLT release` in less than 160 timer time  -> treated as typo and send default layer keys.   
 
The code is not optimized nor bug free and not well tuned, but it's in working condition:https://github.com/hiro2016/ergodox_gui_configurator/blob/master/qmk_firmware/delayed_lt.c  

How to use:

 - Download the file above.
 - Place the fine in qmk root dir
 - #include  "delayed_lt.c"
 - Insert ` if(!process_action_delayed_lt(keycode,record)) return false;` to your process_record_user.  

Bugs:  
Not handling shift or other modifiers in toggled layers as of now.     
Not optimized code.  Since idk nothing of qmk, there are probably unnecessary field variables added.  
Tapping Term is hard coded to 160.   
Only supports up to layer 7.  
Tap count etc are not handled. Right now, my code is basically a big marco not integrated into qmk nor using qmk functionalities.  
  
I wil very much appreciate if someone could tell me anyone of the below:  

### On my JIS keyboard is sending,  `0001001000100100` and  `0001001000011111` for LSHFTED or RSHFTED 7  and 2 respectively.  I believe upper 8 bits are the code for shift:

    >>>int('00010010',2)
    18

But I do not recognize the mask....what's the MACRO name for this?   Can anyone point me to qmk code I can handle this sort of things?   
  
### I implemented a custom action code as below and am using bits from 10 to 13 to send layer param:   

    enum macro_keycodes {
      DLT_LAYER_TAP = 0x9100, 
      DLT_LAYER_MAX = 0x9f00,
    };
    #define DLT(layer,kc) (kc | DLT_LAYER_TAP | ((layer & 0x7) << 9))
    #define GET_DLT_LAYER(kc) ((kc >> 9) & 0x7)  
  
process_record_user's parameter, keycode, does not contain upper 8 bits of my custom action code.  
As result, I'm calling `keymap_key_to_keycode(layer, keypos_t);`. Why is this happening?

drashna, 2018-03-25:
LT's speed is determined by TAPPING_TERM, so it can be sped up. 

Also, there are a number of config.h settings that can influence its behavior.
https://docs.qmk.fm/config_options.html

fredizzimo, 2018-04-07:
`LT` currently lacks some control compared to the `MT` for example. This will hopefully be fixed very soon, and after that you should be able to control it to get the behaviour you want.

closed 2018-04-07 by hiro2016

hiro2016, 2018-04-07:
Thanks for the comments.  
I have tested some ideas and gave up writing them as action codes long time ago, so closing this issue.
  
>LT's speed is determined by TAPPING_TERM, so it can be sped up.  

Yes, but I feel a single TAPPING_TERM for all keys is wrong. TAPPING_TERM right for `E` key is wrong for `F` key and having to make sure LT key is held down while other keys are tapped does not contributes to a comfortable typing experience. Now I have given up on my DLT idea as an action code.      
  
In the end I wrote code that more or less:  
+ can be called from custom function or macro; so a variable that serves as tapping_term can be set there.  
+ toggles on buffering/filtering of all user inputs at key down time  
+ evaluates user inputs and decide what to do at key release time  
    
Since a combo input function of a similar design which I wrote later is working fine, the design is functional I believe;DLT itself has been abandoned for a long time now.

fredizzimo, 2018-04-07:
Another change that I'm planning for tapping keys, is to provide custom settings, like `TAPPING_TERM` for every key separately. So once that's implemented, it will hopefully be enough for your needs, or at least one step closer.

hiro2016, 2018-04-08:
@fredizzimo  
>Another change that I'm planning for tapping keys, is to provide custom settings, like TAPPING_TERM for every key separately. So once that's implemented, it will hopefully be enough for your needs, or at least one step closer.    

Wow, that truly is an improvement.     
But I must warn you that:  
+ Most people don't know the right tapping term for each finger, so they may not be able to fine tune them.  
+ without an option to disable LT etc for the same keyboard half, with TAPPING_TERM  set to a fairly low value, roll over typing becomes difficult.    
+ Apart from Index finger home low keys like `F` and `J`, other fingers/keys require relatively high and similar TAPPING_TERMs, so the usefulness of the option might be limited.  
  

#### Low LT TAPPING_TERM even for a single key and roll over typing don't get along quite well.    
My case, learning not to do roll over with F and J was not hard at all; especially with my custom LT function disabled for the same keyboard half. But casual users would likely to refuse changing their typing habit, so introducing variable TAPPING_TERM for each key would mean only reducing LT threshold time by like 30ms for some keys by factory default;still a rather noticeable difference though.   
  
#### In the end, combo input might be better than LT  
Faster smoother typing experience for LT is capped by the habit of rolling over. From the keyboard side, detecting whether the user is requesting for LT action performed or just rolling over is hard to tell when TAPPING_TERM is set low.     
  
On the other hand, combo inputs always have a very detectable signature; two keys presses within a 50ms interval almost never happens when I'm typing normally.     
   

Having said all the above, enthusiasts like myself would love variable TAPPING_TERM for each column/key as it makes using LT a lot more comfortable.