qmk/qmk_firmware #1710: [Discussion] Make arguments redo

jackhumbert, 2017-09-13:
We're currently considering revamping the make system a bit to support organising keyboards/projects into folders, and letting those folders share code with each other. This will replace what's known as "subprojects" right now, and would support up to 5 folders in depth. One example might be:

    /keyboards/olkb/planck/rev4/

The default keymap for which could be compiled with:

    make olkb/planck/rev4:default

Changing the separator to `:` allows for more clarity, and more easily supports `-` in folder/keymap names. Other targets would be separated with `:` as well, like `make olkb/planck/rev4:default:dfu`.

You would be able to define a default at any point in the folder system, so `olkb/rules.mk` could have:

    DEFAULT_FOLDER = olkb/planck/rev4

so that whenever `make olkb:default` is run, it defaults to the `olkb/planck/rev4` target - this would be required (this is up for debate, but there are some unwanted consequences with building all available folders).

The `*.c` files would be included automatically if they exist. For example:

* /keyboards/olkb/olkb.c
* /keyboards/olkb/planck/planck.c
* /keyboards/olkb/planck/rev4/rev4.c

would all get included when compiling the target `olkb/planck/rev4`, and each folder would get added to the path.

Filenames would convert `/` to `_` in the target.

Keymaps could be stored at any point in the folder system, allowing for keyboard-specific keymaps, revision-specific keymaps, etc.

I've been messing around with this on [make_args_redo](https://github.com/qmk/qmk_firmware/commits/make_args_redo) but we're welcoming comments on the concept/implementation!

jetpacktuxedo, 2017-09-18:
It might be a struggle for people used to the current system, but it should make it easier for new people and for future changes.

jackhumbert, 2017-09-25:
I think we'll plan on implementing this on **2017-10-01** - if anyone has an ideas or concerns with the concept or implementation, please comment before then!

Sebb767, 2017-10-14:
Looks good. I like that subprojects are better split from other parameters such as keymap and programmer than before. +1 from me :)

closed 2017-10-15 by jackhumbert