qmk/qmk_firmware #1716: RGui missing on Ubuntu when typing in fi_FI locale in Ubuntu

rask, 2017-09-15:
I'm posting here as I'm not certain if this could be a firmware problem or not. I have a problem with the RGui (Super_R) key in Ubuntu 14.04.

I recently compiled a firmware for the UK78 keyboard with a `1.5u 1.5u 7u 1.5u 1.5u` bottom row layout, with keys laid out as `[LCtrl][LAlt][Space][Ralt][RGui]`.

Now when I type in US ANSI (en_US) layout everything works as it should, with the RGui acting as Super normally.

When I switch to ISO Nordic (fi_FI) as the typing language the RGui key just disappears completely. Switching back to US ANSI restores it.

When I check out input with `xev` I can see the RGui working in US ANSI but in ISO Nordic even xev does not register any key presses.

`xmodmap` shows Super_R in place on US ANSI, but it is missing on ISO Nordic.

Any ideas if this is an issue with my firmware or if this is outside the domain of QMK? The keymap I flashed into UK78 is the `default/keymap.c` with some keys laid out differently.

jackhumbert, 2017-09-16:
Does ISO Nordic use RGUI as a dead key or anything? That's the only thing I can thing of. If ANSI is receiving the right code, I'd suspect it's something with the OS.

fredizzimo, 2017-09-19:
To me it seems to be a problem with the keymap on the Ubuntu side. 

But what do you need RGUI for? Many normal keyboards lack that key, could you use LGUI instead?

rask, 2017-09-19:
It seems the UK78 has some bootmagic option enabled, so I will need to try with the updated UK78 base and see if that fixes the issue.

@jackhumbert not certain, never used a QMK ISO Nordic, just stock ones.

@fredizzimo I'm thinking of swapping the RGui for LGui just to keep it working in case the UK78 reflash without bootmagic does not work. :)

drashna, 2018-03-25:
Bootmagic strikes again.
https://docs.qmk.fm/feature_bootmagic.html#bootmagic-keycodes

Easiest option is to reset eeprom.  If bootmagic is enabled, hold space and backspace when plugging it in.