qmk/qmk_firmware #1730: LT with Mod,Key

yanfali, 2017-09-18:
I'm trying to do this:

LT(1, S(LGUI(KC_R))) but it only sends KC_R and the LT doesn't seem to work.

If I do S(LGUI(KC_R)) works fine. Is this a bug or just unsupported? Thanks. QMK rocks.

hiro2016, 2017-09-19:
#define LGUI(kc) (kc | QK_LGUI)
    QK_LGUI               = 0x0800,
    #define LT(layer, kc) (kc | QK_LAYER_TAP | ((layer & 0xF) << 8))
    QK_LAYER_TAP          = 0x4000,
    QK_LAYER_TAP_MAX      = 0x4FFF,  
    #QK_LAYER_TAP          = 0x4000

Here's binary representations of the above numbers.

    "{0:08b}".format(0x4000)
    '100000000000000'  
    >>> "{0:08b}".format(0x4fff)
    '100111111111111'  
    # QK_LGUI               = 0x0800,
    "{0:08b}".format(0x0800)
    '100000000000'

Any keycode is an uint16_t, and lower 8 bits of keycode is used to hold hid usage id.   
In the case of LT, it uses 9th to 14th bits to send layer param.  
QK_GUI has a bit set at 12th position, so unless QMK people has used some wonderful trick which I don't know, what you tried is not supported.    
  
You should try using a macro like below:  
  
    register_code(KC_LGUI);
    register_code(KC_RSHIFT);
    register_code(KC_R);
    unregister_code(KC_R);
    unregister_code(KC_RSHIFT);
    unregister_code(KC_LGUI);  

I have not really looked into qmk yet, so I might be wrong though.

yanfali, 2017-09-19:
@hiro2016 thanks, that's very helpful. I will give the macro approach a try.

yanfali, 2017-09-19:
I was able to define the Macro, but sadly it doesn't work with layer tap. Now I have to look into how layer tap is implemented.

```c
const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt) {

  switch(id) {
    case 0:
      if (record->event.pressed) {
		  return MACRO(D(LSFT), D(LGUI), T(R), END);
	  }
      else {
		  return MACRO(U(LSFT), U(LGUI), END);
	  }
      break;
  }

  return MACRO_NONE;
};
```

yanfali, 2017-09-19:
Anyone know how MACROTAP works?

hiro2016, 2017-09-19:
Ya, you wanted LT..  
I was half asleep ... hmm, maybe you have to write a custom keycode and add new functions to `process_record` etc.

<code>
uint16_t custom_lt_timer;
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    if(keycode == custom_keycode){
    if(record->event.pressed){
        custom_lt_timer = timer_read();
        layer_on(4);
    }else{
        layer_off(4);
        if(timer_elapsed(custom_lt_timer)<200){ 
            register_code(KC_LGUI);
            register_code(KC_RSHIFT);
            register_code(KC_R);
            unregister_code(KC_R);
            unregister_code(KC_RSHIFT);
            unregister_code(KC_LGUI);  
        }
    }}
}</code>

It's only a pseudo code but you could do something like the above.

yanfali, 2017-09-19:
Neat. Thanks, I will give it a shot this evening.

yanfali, 2017-09-20:
Thanks @hiro2016 that worked.

```c
// Hard Reload Chrome
enum dz60_keycodes {
  LT_1_OR_RELOAD_CHROME = SAFE_RANGE
};

uint16_t custom_lt_timer;
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
	if(keycode == LT_1_OR_RELOAD_CHROME) {
		if(record->event.pressed) {
			custom_lt_timer = timer_read();
			layer_on(1);
		}else{
			layer_off(1);
			if (timer_elapsed(custom_lt_timer)<200) {
				register_code(KC_LGUI);
				register_code(KC_RSHIFT);
				register_code(KC_R);
				unregister_code(KC_R);
				unregister_code(KC_RSHIFT);
				unregister_code(KC_LGUI);
			}
		}
	}
	return true;
}
```

Does  this look right to you?

yanfali, 2017-09-21:
OK, since this is actually working and about to get merged I'm going to close this. Thanks @hiro2016

closed 2017-09-21 by yanfali

hiro2016, 2017-09-21:
Congrats  
Your code looks beautiful.  
  
I'm glad I was some kind of help.

yanfali, 2017-09-21:
LOL, since it was your code, you deserve all the credit. Thanks again, I wouldn't have been able to figure this out as a easily without your help.