qmk/qmk_firmware #1732: error: enumeration value 'USB_EVENT_UNCONFIGURED' not handled in switch

olivierpons, 2017-09-18:
Hi all,

After managing to solve the problem of all dependencies (just read the .gitmodule and "git clone"), i cant compile ergodox_infinity.

When I try with "make ergodox_infinity-all" I get this error for each try:

```
tmk_core/rules.mk:358: recipe for target '.build/obj_ergodox_infinity_yoruian/protocol/chibios/usb_main.o' failed
make[1]: *** [.build/obj_ergodox_infinity_yoruian/protocol/chibios/usb_main.o] Error 1
Making ergodox_infinity with keymap zweihander-osx and target all                              [ERRORS]
./tmk_core/protocol/chibios/usb_main.c: In function 'usb_event_cb':
./tmk_core/protocol/chibios/usb_main.c:771:3: error: enumeration value 'USB_EVENT_UNCONFIGURED' not handled in switch [-Werror=switch]
   switch(event) {
   ^
cc1: all warnings being treated as errors

```

fredizzimo, 2017-09-19:
This is most likely caused by the wrong version of the Chibios submodule being checked out.

Could you provide the output of `git status`?

olivierpons, 2017-09-20:
You're right, I just had to do a `git clone --recursive https://github.com/kiibohd/controller.git` but I didn't find that in the documentation. Thank you very much for your help!

closed 2017-09-20 by olivierpons