qmk/qmk_firmware #1738: Infinity Ergodox support

olivierpons, 2017-09-20:
According to [Infinity Ergodox FAQ](https://input.club/knowledgebase/infinity-ergodox-faq/), the Infinity ErgoDox Kit uses the MK20DX256VLH7 microchip.

If I try to compile with the latest QMK firmware like this:
```
git clone --recursive https://github.com/qmk/qmk_firmware.git
cd qmk_firmware
make ergodox_infinity-default
```
Everything runs smoothly (very nice).
But when I try to put the code into the Infinity Ergodox, I get this:
```
sudo dfu-programmer mk20dx256vlh7 flash ergodox_infinity_default.hex
[sudo] password for olivier: 
Unsupported target 'mk20dx256vlh7'.
targets:
    at89c51snd1c       at89c51snd2c       at89c5130          at89c5131       
    at89c5132          at90usb1287        at90usb1286        at90usb1287-4k  
    at90usb1286-4k     at90usb647         at90usb646         at90usb162      
    at90usb82          atmega32u6         atmega32u4         atmega32u2      
    atmega16u4         atmega16u2         atmega8u2          at32uc3a0128    
    at32uc3a1128       at32uc3a0256       at32uc3a1256       at32uc3a0512    
    at32uc3a1512       at32uc3a0512es     at32uc3a1512es     at32uc3a364     
    at32uc3a364s       at32uc3a3128       at32uc3a3128s      at32uc3a3256    
    at32uc3a3256s      at32uc3a4256s      at32uc3b064        at32uc3b164     
    at32uc3b0128       at32uc3b1128       at32uc3b0256       at32uc3b1256    
    at32uc3b0256es     at32uc3b1256es     at32uc3b0512       at32uc3b1512    
    at32uc3c064        at32uc3c0128       at32uc3c0256       at32uc3c0512    
    at32uc3c164        at32uc3c1128       at32uc3c1256       at32uc3c1512    
    at32uc3c264        at32uc3c2128       at32uc3c2256       at32uc3c2512    
    atxmega64a1u       atxmega128a1u      atxmega64a3u       atxmega128a3u   
    atxmega192a3u      atxmega256a3u      atxmega16a4u       atxmega32a4u    
    atxmega64a4u       atxmega128a4u      atxmega256a3bu     atxmega64b1     
    atxmega128b1       atxmega64b3        atxmega128b3       atxmega64c3     
    atxmega128c3       atxmega256c3       atxmega384c3    
```
Is there a plan for supporting `mk20dx256vlh7` microchip or am I missing something?

fredizzimo, 2017-09-20:
You need to use `dfu-util` not `dfu-programmer` for flashing the Infinity Ergodox, so install that with `apt-get` first.

The easiest way to flash, unless you have special needs is directly through make `make ergodox_infinity-default-dfu-util` 

If you do it manually, you need the bin file, not the hex, which you can make by `make ergodox_infinity-default-bin`

olivierpons, 2017-09-20:
Omg this is working!

And it seems it's working more nicely than the official kiibohd/controller:
- this one is still buggy (I've complained enough about that everywhere else I think)
- here the LCD is swapping between 2 images : "Num" and another one with the title "1=on D=default B=both".

Is there any documentation for:
- changing the color of the LCD,
- customizing the LCD,
- making macros for the Ergodox

?
Thank you very much, you've put me on the right track I just hope to go further from now on!

fredizzimo, 2017-09-20:
The master (the keyboard connected to the computer) will make most decisions, the slave will mostly transfer the physical state of the buttons. In many cases it's enough to flash the master only, but not in all, so it's recommended to always flash both halves with the same firmware. 

The keyboards themselves don't know which half they are, so you need to specify that separately, more instructions can be found [here](https://github.com/qmk/qmk_firmware/tree/master/keyboards/ergodox_infinity)

For macros and stuff you can read the [general QMK documentation](https://docs.qmk.fm/)

For the LCD customization there's unfortunately no documentation right now, but I recommend to implement a `simple_visualizer` like [this](https://github.com/qmk/qmk_firmware/blob/master/layouts/community/ergodox/333fred/visualizer.c). Note that the colors are specified in the `HSI` color space with ranges from 0-255, so it's not RGB.

It's also possible to customize the visualizer almost completely, look at how the [simple visualizer itself is implemented](https://github.com/qmk/qmk_firmware/blob/master/keyboards/ergodox_infinity/simple_visualizer.h) for example, or the [default visualizer](https://github.com/qmk/qmk_firmware/blob/master/keyboards/ergodox_infinity/visualizer.c)

In all cases, you can just create a `visualizer.c` file in your layout folder. And add your implementation there. You can start by copying any of those existing visualizers.

olivierpons, 2017-09-21:
Thanks a lot! You're very kind and helpful. I open another issue where I'm trying to get more "clues" on how to fully (and properly) customize the lighting of my Infinity Ergodox.

closed 2017-09-21 by olivierpons