qmk/qmk_firmware #1739: Macro speed

rasmusx, 2017-09-20:
Hello,

I am currently porting FC660C Hasu controler to QMK. Everything seems to be working, but there is one problem.

QMK macro speed is a lot slower compared to TMK. This goes for both SEND_STRING and MACRO with interval set to 0. Any reason why it could be like this?

I will try to use some other board tonight with QMK and compare the macro speed.

fredizzimo, 2017-09-20:
How long strings are you sending? There is indeed some unnecessary overhead in those functions, but unless your string are very long, it should just be a few milliseconds.

rasmusx, 2017-09-21:
I made a test.

I created a script to help me measure running time of macro.
https://gist.github.com/rasmusx/0bf2f4b998d4715d6c90385f69ed639b

Macro i used to run looks like this:
```
MACRO( \
  T(ENT), D(LCTRL), D(LSHIFT), T(U), U(LCTRL), U(LSHIFT), \
  T(0),  T(0),  T(2),  T(8), \
  D(LCTRL), D(LSHIFT), T(U), U(LCTRL), U(LSHIFT), \
  T(2),  T(5),  T(6),  T(F), \
  D(LCTRL), D(LSHIFT), T(U), U(LCTRL), U(LSHIFT), \
  T(0),  T(0),  T(B),  T(0), \
  D(LCTRL), D(LSHIFT), T(U), U(LCTRL), U(LSHIFT), \
  T(2),  T(5),  T(A),  T(1), \
  D(LCTRL), D(LSHIFT), T(U), U(LCTRL), U(LSHIFT), \
  T(0),  T(0),  T(B),  T(0), \
  D(LCTRL), D(LSHIFT), T(U), U(LCTRL), U(LSHIFT), \
  T(F),  T(F),  T(0),  T(9), \
  D(LCTRL), D(LSHIFT), T(U), U(LCTRL), U(LSHIFT), \
  T(2),  T(5),  T(6),  T(F), \
  D(LCTRL), D(LSHIFT), T(U), U(LCTRL), U(LSHIFT), \
  T(F),  T(E),  T(3),  T(5), \
  D(LCTRL), D(LSHIFT), T(U), U(LCTRL), U(LSHIFT), \
  T(0),  T(0),  T(2),  T(0), \
  D(LCTRL), D(LSHIFT), T(U), U(LCTRL), U(LSHIFT), \
  T(2),  T(5),  T(3),  T(B), \
  D(LCTRL), D(LSHIFT), T(U), U(LCTRL), U(LSHIFT), \
  T(2),  T(5),  T(0),  T(1), \
  D(LCTRL), D(LSHIFT), T(U), U(LCTRL), U(LSHIFT), \
  T(2),  T(5),  T(3),  T(B), \
  T(SPACE), T(ENT), END \
)
```
Output of macro: ```(╯°□°）╯︵ ┻━┻```

Running time with QMK: **1.38 seconds**
Running time with TMK: **0.17 seconds**

Unfortunately I haven't had the chance to test other keyboard firmware yet.

jackhumbert, 2017-09-21:
This is pretty interesting, and that macro describes my feelings on this pretty well :/

I've done a quick look into the chain of calls for the macro stuff, and I couldn't find anything that we've changed, but I'd definitely like to look into this more.

rasmusx, 2017-09-21:
Yeah its pretty interesting.

I did quick test on different PC while using Filco keyboard with Pegasushoof controller. Results show that with this controller both TMK and QMK are slow to print out macros. Time was around 1.3 seconds.

rasmusx, 2017-10-22:
I have found out that when enabling NKRO macros also run fast. Same with every controller/keyboard.

- slow - NKRO disabled
- fast  - NKRO enabled

closed 2017-10-23 by rasmusx