qmk/qmk_firmware #1743: Backlight as layer feedback help

loadedsith, 2017-09-20:
Hi there.

I had the thought that I could program my keyboard, a planck rv3, could change its backlight level based on which layer is active. That is, when on the default layer, have the backlight set to 5, but when on the lower layer turn the backlight off, and when on the raise layer to set the backlight to 10.

I've tried a few things, first I tried [this in process_record_user](https://gist.github.com/loadedsith/2cb23ade3911636637bb7700b39633e5). Which set the backlight correctly on press, but to failed to reset the backlight on the first release, but when pressing and releasing the key again, it would reset on the release.

Then I tried [this with matrix_scan_user](https://gist.github.com/loadedsith/0eb4049bde4b519d6e628d647d5226e6). I have a harder time explaining how this failed, but it seemed to sometimes flicker between lower and default's backlight levels.

I was hoping someone out there had done something similar or could explain what I'm not understanding about these 2 functions.

Thanks!

drashna, 2017-10-03:
I'm not sure how much of this would carry over, but....

Specifically, I do this with the RGB Underglow on the Ergodox EZ.   But I do this by detecting which layer it is on (matrix_scan_users), but since the RGB Light commands are resource intensive, I only set it when the layer has changed.  Once it's changed, it doesn't need to be called again.

https://github.com/qmk/qmk_firmware/blob/master/keyboards/ergodox_ez/keymaps/drashna/keymap.c

If the backlight works the same (only needs to be called once), then this is what you'd want to do, as well.  Add the following to the matrix_scan_user(void) function: 

`    static bool has_layer_changed = true;
    uint8_t new_layer = biton32(layer_state);
    static uint8_t old_layer = 0;
    if (old_layer != new_layer) {
        has_layer_changed = true; 
        old_layer = new_layer;
    }
    if (has_layer_changed) {
        switch (new_layer) {
            case 1:
                //backlight command for layer 1;
                break;
            case 2:
                //backlight command for layer 2;
                break;
            case 3:
                //backlight command for layer 3;
                break;
            default:
                //backlight command for default layer (0);
                break;
        }
        has_layer_changed = false;
    }`

Setting the "has_layer_changed" boolean to true by default ensures that it will run the backlight command at least once. 

Otherwise, you may want to add the command to "matrix_init_user" as this runs startup commands.

loadedsith, 2017-10-03:
Thanks @drashna. I've got a working [snippet here](https://gist.github.com/loadedsith/99d79b19cf392396cd62e804e3c4cd6a).

closed 2017-10-03 by loadedsith