qmk/qmk_firmware #1747: Infinity Ergodox backlighting intensity not to 100%

olivierpons, 2017-09-21:
Hi,
I'm trying to make my own 100% customized `azerty` Infinity Ergodox keymap _and_ lighting in the `community` folder.

I want the LCD to be flashy = maximum intensity. I want the base layer to be full white, full intensity. According to what I've found, and with the very nice help of @fredizzimo , the `RGB(255,255,255)` should be converted to `LCD_COLOR(0, 0, 255);`.
But when I plug my Ergodox, I can clearly see a full blowing pure white, then it goes down to almost grey, and stays always "grey" = very "low" white. I guess it's all about the intensity.

So I've add the code `state->target_lcd_color = LCD_COLOR(0, 0, 255);` (see after) and the base layer is white but not "glowy" at all.

I've found out somewhere in the code stuff with `lcd_backlight_brightness(255);`. It's not working.
Another thing is `lcd_backlight_hal_color(65535,65535,65535);`. This one is working! Wow!
So I've tried to add this line in the beginning of `get_visualizer_layer_and_color()` then I get a strange behavior: it's "flashing" white each time I press a key then goes instant to very "low" white. It feels like somewhere in a loop, when the state of the key changes, and only when it changes, that it calls `get_visualizer_layer_and_color()`.

My question is: how to completely override both backlighting handling and LCD writing handling?

Here's what I've done so far:

- I've made my folder `./layouts/community/ergodox/azerty/`
- I've copied `default/visualizer.c` to my folder `azerty`
- I've copied `default/keymap.c` to my folder `azerty`
- I've copied `default/rules.mk` to my folder `azerty`

Here's the content of `rules.mk`: 
```
LCD_BACKLIGHT_ENABLE = yes                                                       
LCD_ENABLE = yes                                                                 
BACKLIGHT_ENABLE = yes                                                           
NKRO_ENABLE = yes                                                                
KEY_LOCK_ENABLE = yes    
```
I've modified `visualizer.c` to try to make the backlight to it white maximum intensity, without success. Here's the file, with my minor modifications, and notes in the comment to show what I've found so far:

```
/*                                                                               
Note: this is a modified copy of ../default/visualizer.c, originally licensed GPL.
*/                                                                               
                                                                                 
#include "simple_visualizer.h"                                                   
                                                                                 
// This function should be implemented by the keymap visualizer                  
// Don't change anything else than state->target_lcd_color and state->layer_text as that's the only thing
// that the simple_visualizer assumes that you are updating                      
// Also make sure that the buffer passed to state->layer_text remains valid until the previous animation is
// stopped. This can be done by either double buffering it or by using constant strings
static void get_visualizer_layer_and_color(visualizer_state_t* state) {          
    // LCD_COLOR(  0,   0, 255); // White  <> RGB(255,255,255)                   
    // LCD_COLOR( 85, 255, 255); // Green  <> RGB(  0,255,255)                   
    // LCD_COLOR(  0, 255, 255); // Red    <> RGB(255,  0,255)                   
    // LCD_COLOR(170, 255, 255); // Blue   <> RGB(  0,  0,255)                   
    // LCD_COLOR(213, 255, 255); // Purple <> RGB(255,  0,255)                   
    // LCD_COLOR( 42, 255, 255); // Yellow <> RGB(255,  0,255)                   
    // LCD_COLOR(128, 255, 255); // Aqua   <> RGB(  0,255,255)                   
    if (state->status.layer & 0x10) {
        state->target_lcd_color = LCD_COLOR(140, 100, 60);
        state->layer_text = "Not used WTF";
    } else if (state->status.layer & 0x8) {
        state->target_lcd_color = LCD_COLOR(168, 255, 0xFF);
        state->layer_text = "Numpad mode";
    } else if (state->status.layer & 0x4) {
        state->layer_text = "Cursor";
        state->target_lcd_color = LCD_COLOR(255, 255, 0xFF);
    } else if (state->status.layer & 0x2) {
        state->layer_text = "Code";
        state->target_lcd_color = LCD_COLOR(84, 255, 0xFF);
    } else {
        state->layer_text = "Couche de base";
        state->target_lcd_color = LCD_COLOR(0, 0, 255);
    }                                                                            
} 
```

olivierpons, 2017-09-21:
If you remove some files from your folder, then the compiler will take the default files.
This is nice, so you just have to copy/paste the right files, and you override the default files.
But where are the defaults files? Is there an option to display this?

And if I remove my files in my folder `azerty/simple_visualizer.h` and  `azerty/visualizer.c`, then my Infinity Ergodox goes back to the default animation (which is nice). But where are those files?

Is there a way to debug? Send printf to the Linux console or whatever?

rasmusx, 2017-09-21:
@olivierpons  use hid_listen for debug.

olivierpons, 2017-09-21:
Thank you for your suggestion, but may I ask how you use that? Any link or sample on how to debug/print screen?

drashna, 2018-03-25:
This may help: 
https://docs.qmk.fm/newbs_testing_debugging.html

fredizzimo, 2018-04-07:
You should be able to control the brightness with `lcd_backlight_brightness`, it takes a value from 0-255. The plan was to be able to control that, and save the settings to EEPROM, like you can for backlight levels, but it has never been implemented.

closed 2019-11-21 by drashna