qmk/qmk_firmware #1756: [Request] logarithmic backlight control

ivancuric, 2017-09-23:
How would one go about implementing a logarithmic brightness control on an ATmega32u4?
I've fiddled on my Arduino a bit, and came up with the following:
```c
    mapped = map(count, 0, steps, 0, 100);
    reduced = mapped * 0.01;
    eased = (pow(2, reduced) - 1) * 100;

    ledmapped = map(eased, 0, 100, 0, 255);
    analogWrite(ledPin, count);
```
Basically, mapping 100 steps to 255, but on a logarithmic scale, so that the brightness has a perceptively linear scale. The current implementation is linear, so the brightness seems to grow exponentially.

I found the code responsible for the backlight in `quantum.c`:
https://github.com/qmk/qmk_firmware/blob/master/quantum/quantum.c#L824

However, I can't make any sense of it - too low level for my basic C knowledge.

Does anyone have any ideas?

ivancuric, 2017-09-24:
Did some more work, implemented in C:
```c
#include <stdint.h>
#include <stdio.h>

#define BACKLIGHT_LEVELS 8

void main()
{
    uint16_t i;
    uint16_t orig(uint16_t level) {
        if ( level == 0 ) {
            return 0;
        } else if (level == BACKLIGHT_LEVELS) {
            return 0xFFFF;
        }
        return 0xFFFF >> ((BACKLIGHT_LEVELS - level) * ((BACKLIGHT_LEVELS + 1) / 2));
    }

    uint16_t new(uint16_t level) {
        return ((level * level * 0xFFFF) - 0xFFFF) / (BACKLIGHT_LEVELS * BACKLIGHT_LEVELS - 1);
    }

    printf("Orig\tNew\n-------------\n");
    for (i = 0; i <= BACKLIGHT_LEVELS; i++)
    {
        printf("%d\t", orig(i));
        printf("%d\n", new(i));
    }
}
```
Output:
```
Orig    New
-------------
0       fbf0
0       0
0       c30
0       2081
0       3cf3
f       6185
ff      8e38
fff     c30b
ffff    ffff
```

I can't math today, I have the dumb.
Any ideas how to fix the new output to go from `0x0` to `0xffff` ?

AleksandarDev, 2017-09-24:
The formula should be: `(x^2)*(map_max/(map_steps^2))` where `map_max` is max value required, `map_steps` is number of horizontal steps you have.

You can optimize this equation by caching/calculating `(map_max/(map_steps^2))` value (since these never change)

----

`return (level*level)*(0xFFFF/(BACKLIGHT_LEVELS*BACKLIGHT_LEVELS));` 
which is `x^2` mapped from `0 .. 8` to `0x0000 - 0xFFFF` (including edge values)

will give you following [response](https://www.graphsketch.com/?eqn1_color=1&eqn1_eqn=(x*x)*(65535%2F(8*8))&eqn2_color=2&eqn2_eqn=&eqn3_color=3&eqn3_eqn=&eqn4_color=4&eqn4_eqn=&eqn5_color=5&eqn5_eqn=&eqn6_color=6&eqn6_eqn=&x_min=0&x_max=8&y_min=0&y_max=65535&x_tick=1&y_tick=1000&x_label_freq=5&y_label_freq=5&do_grid=0&do_grid=1&bold_labeled_lines=0&bold_labeled_lines=1&line_width=4&image_w=850&image_h=525).

If you want the curve to be steeper, just increment exponential 
eg. `(level*level)*(0xFFFF/(BACKLIGHT_LEVELS*BACKLIGHT_LEVELS*BACKLIGHT_LEVELS)) ` 
which is `x^3` mapped to `0x0000 - 0xFFFF`

@wedranb

ivancuric, 2017-09-24:
Thanks for the response! However the function gives out the final value as `0xFFC0` instead of `0xFFFF`.

AleksandarDev, 2017-09-24:
@ivancuric It's because you're clipping remainder from division (65535 is not dividable by 9 without reminder). Check 3rd paragraph for correct solution (rearranged brackets)

ivancuric, 2017-09-24:
Ah, got it. Thanks a lot.
Sticking with the lookup table tho. 
```c
OCR1x = (uint16_t)((level * level * 0xFFFF) / (BACKLIGHT_LEVELS * BACKLIGHT_LEVELS));
```
Gives good results in the console (`0, 3FF, FFF, 23FF, 3FFF, 63FF, 8FFF, C3FF, FFFF`), but for some reason only works on 3 values when programming the board (0, 1, max)

drashna, 2018-03-25:
Ever finish this?

ivancuric, 2018-03-25:
Nope, but @BalzGuenat and @jackhumbert fixed it!
https://github.com/qmk/qmk_firmware/blob/master/quantum/quantum.c#L933
https://github.com/qmk/qmk_firmware/commit/4931510ad38aadb1769c9241bfad0c3d77ad687f

closed 2018-03-25 by ivancuric

drashna, 2018-03-25:
Ah, glad to hear it then! 
Specifically, fixed by #2187 (tagging for reference)