qmk/qmk_firmware #1757: [Help] Pulsing pin

kalvinpearce, 2017-09-24:
While trying to send a signal from one board loaded with QMK to an arduino reading it on the other end, it seems that setting the pin high works fine but setting it low pulses from low to high to low.

Essentially I am wanting to send an analog signal from my qmk board to a pin on an arduino board. I thought I would start with trying to get high and low working before trying analog values but ended up stumbling upon this.

Basically the only custom code is in process_record_user
`bool process_record_user(uint16_t keycode, keyrecord_t *record) 
{	
	if (record->event.pressed) 
	{
		switch(keycode) 
		{
			case KC_LSFT:
				keyComboState |= 0b00000001;
				break;
			case KC_LCTL:
				keyComboState |= 0b00000010;
				break;
            case KC_DOWN:
				keyComboState |= 0b00000100;
				break;
        }
	}
	else
	{
		switch(keycode) 
		{
            case KC_LSFT:
				keyComboState &= ~(0b00000001);
				break;
			case KC_LCTL:
				keyComboState &= ~(0b00000010);
				break;
			case KC_DOWN:
				keyComboState &= ~(0b00000100);
				break;
        }
	}

	if( keyComboState == 0b00000111 )
	{
		PORTF |= 0b00000001;
		return false;
	}
	else
	{
		PORTF &= ~(0b00000001);
	}

	return true;
}`

fredizzimo, 2017-09-29:
Did you configure the arduino input wit a pull up or without?

kalvinpearce, 2017-09-30:
I have tried both. Without pullup it pulses, with pullup it sits almost at full high. It doesn't seem to go to ground

fredizzimo, 2017-09-30:
Hm, are you that you have configured the pin as outpu on the QMK side, and that the matrix scan or something doesn't touch it.

seebs, 2017-11-22:
Does DDRF have 0b00000001 set?

Is anything else setting PORTF perhaps? A lot of time, ports are being used for other I/O, and sometimes code will just set a whole port register even though it only thinks 6 of the bits are in use, because it assumes the other 2 bits don't matter.

drashna, 2018-03-25:
were you able to resolve this?