qmk/qmk_firmware #1761: Difficulty writing macro due to avrc and teansy particularities.

hiro2016, 2017-09-25:
I sort of feel this is not the place to post this.  
There is nowhere else, however, and since other tinkers are likely to face a similar issue, decided to post here.  

## Background 
I have to used about 88 macros because one keyboard I'm trying to mimic maps about 30 keys to 90 literals using right and left shifts.  Non-Latin languages have too many literals indeed.   
  
## Problem I faced
When I created  71 or so macros, my pjrc teansy loader stated the hex file is too big.    
  
## Cause
Inline function.   
Normally compilers know when to embed and when not to better than I, but this time around the compiler did not take teensy2's rom size into account.  

The inline function I had:  

    void inline _send_key(uint32_t layer, keypos_t key ){
     uint16_t keycode = keymap_key_to_keycode(layer,key);
          if( (QK_MACRO <= keycode) && (keycode <QK_MACRO_MAX) ){
              action_t action;
              keyrecord_t record;
              action.code = keycode;
              record.event.key = prv_keypos;
              record.event.pressed = true;
              record.event.time = timer_read();
              action_get_macro(&record,action.func.id, action.func.id);
              record.event.pressed = false;
              action_get_macro(&record,action.func.id, action.func.id);
              return;
          }
          switch(keycode>>8){
            case(0x2):
              register_code(KC_LSHIFT);
              register_code(keycode & 0xff);
              unregister_code(keycode & 0xff);
              unregister_code(KC_LSHIFT);
              break;
            case(0x12):
              register_code(KC_LSHIFT);
              register_code(keycode & 0xff);
              unregister_code(keycode & 0xff);
              unregister_code(KC_LSHIFT);
              break;
            default:
              register_code(keycode & 0xff);
              unregister_code(keycode & 0xff);
          }
        }

The caller function( I have about 70 similar macos):

    case 94:
        //ki
        if(!record->event.pressed){
            if(!clt_pressed){
                register_code(0x0a);
                unregister_code(0x0a);
            }else{  
                // CLT layer switch key was pressed after ki macro keypress
                // but before the ki macro key release. 
                // Running another macro in another layer.
                _send_key(clt_layer,record->event.key);
               clt_pressed = false;
            }
        }
        break;
  
Initially I thought maybe too many conditionals but I'm now suspecting record_t struct definitions in conditionals might have caused function stack size bloating like:  

    switch statement
      if statement
        record_t
      if statement2
        record_t  
      if statement2
        record_t      
     .
     .
     .
     if statement90
       record_t
    
## What I would like to say  
To untrained programmers like myself, this kind of things gives a false impression on the capability of a MPU.  
A word of warning in macro doc would have been nice.   
Or maybe we need a community wiki.

fredizzimo, 2017-09-25:
I'm pretty sure there are ways to do what you are want to do, even with the memory limitations. But I need to see your whole keymap, so that I can get a better overview of what kind of code and data duplication you have there.

However the most obvious thing, is `inline`, which means take this piece of code and paste it inline everywhere the function is called. So it seems that perhaps you have got the meaning backwards? That will cause a lot of code bloat, unless the compiler decides that inlining wasn't really what you wanted, and doesn't inline the function. Inlining can save size when the function is small enough, but typically inlining is done for speed, not for size.

What you need here, is `static` instead of inline, which tells that the function is internal to the compilation unit.

The other easy thing to check is that you don't have any keyboard features enabled that you don't need, so disable any option you don't need in your keymap `rules.mk`, check your keyboard `rules.mk` for a list of features that it enables by default.

hiro2016, 2017-09-25:
Good day fredizzimo  
  
Thanks for your comment.    
I have already got it to work, so no worries.  

 > inline, which means take this piece of code and paste it inline everywhere the function is called    

I guess I did not understand inline keyword well enough.  
I heard increasing function stack size above a certain limit has a negative impact on performance, so I thought:  
 - inline functions are inlined when caller function's stack size is reasonably small  
 - called as a normal function when caller function's stack size is too large   
  
But then, performance and code size is a different issue and avrc compiler supports a wide range of MPUs, so I was a fool to relay on the compiler to do everything for me.   
  
MPU programming is so hard and you guys are amazing for running everything work flawlessly.

closed 2017-09-26 by hiro2016