qmk/qmk_firmware #1766: Difference between Goto and Toggle Layer?

olivierpons, 2017-09-26:
The only documentation about Goto and Toggle I've found is here in the code source `./quantum/quantum_keycodes.h`, but I still dont understand the differences, and when to use the macro `TO()` and when to use `TG()`:

```
// GOTO layer - 16 layers max
// when:
// ON_PRESS    = 1
// ON_RELEASE  = 2
// Unless you have a good reason not to do so, prefer  ON_PRESS (1) as your default.
// In fact, we changed it to assume ON_PRESS for sanity/simplicity. If needed, you can add your own
// keycode modeled after the old version, kept below for this.
/* #define TO(layer, when) (layer | QK_TO | (when << 0x4)) */
#define TO(layer) (layer | QK_TO | (ON_PRESS << 0x4)

// Toggle to layer - 256 layer max
#define TG(layer) (layer | QK_TOGGLE_LAYER)
```

algernon, 2017-09-26:
As far as I understand, `TG` (Toggle) will turn the desired layer on, leaving all the others untouched, while `TO` (Goto) will turn the others off. In practice, this means that if you are on the base layer, and `TG(1)`, and you have a transparent key there, the base layer will be used for it, because `TG(1)` leaves the base layer active too. With `TO(1)`, the base layer turns off, so the transparent key will do nothing.

olivierpons, 2017-09-30:
Ok I see what you mean, thank you for your answer, maybe the documentation is not clear enough here. Thank you again.

closed 2017-09-30 by olivierpons