qmk/qmk_firmware #1772: Dynamic Macros on Ergodox infinity not working

ckaros, 2017-09-28:
When I try to compile ergodox infinity firmware with dynamic macros enabled I get the following error:

`./quantum/dynamic_macro.h: In function 'dynamic_macro_led_blink':`

`./quantum/dynamic_macro.h:53:5: error: implicit declaration of function '_delay_ms' [-Werror=implicit-function-declaration]`
`     _delay_ms(100);`
`     ^`
`cc1: all warnings being treated as errors`

jayliu50, 2017-09-28:
Ooo, nice. Was just about to write up a bug on this one, but you beat me to it. Add this to your rules.mk as a workaround:

```
BACKLIGHT_ENABLE     = no
```

To whomever is resolving this, it would be nice if we could have `BACKLIGHT_ENABLE = no` by default.

ckaros, 2017-09-28:
This does solve the make error.....but it would be nice to have backlight and dynamic macro support together. I have leds installed on my keyboard and they do not turn on with this setting change.

ckaros, 2017-09-28:
Also, now that I have the make issue work around....I am finding that that dynamic macro capability is quite buggy on the ergodox infinity. After recording a macro, if I try to replay, most of the time the keyboard will crash and become unresponsive. I have to unplug and replug to get it to work again. Sometimes it will only type one key then crash.

It's super weird, because sometimes it works, But most of the time it just crashes.

jayliu50, 2017-09-29:
Ah, I see.. Well, you could open up `./quantum/dynamic_macro.h` , and comment out line 53, if you want to keep the backlight enabled.

Yeah, the dynamic macro has either changed in how it's set up, or is plain broken in recent builds. I compiled QMK recently on my original ergodox as well as the infinity, and dynamic macro doesn't work anymore on either of them.

fredizzimo, 2017-09-29:
The problem is the `_delay_ms`, it should be changed to `wait`. Could you test that and send a pull request if it works?

olivierpons, 2017-09-30:
I've compiled on my Infinity Ergodox with my customized macros and it works very well... that's strange.
I've made my own layout, and in the `layouts/community/ergodox/azerty/rules.mk` file I have:

```
LCD_BACKLIGHT_ENABLE = yes                                                       
LCD_ENABLE = yes                                                                 
BACKLIGHT_ENABLE = yes                                                           
BACKLIGHT_BREATHING = yes                                                        
NKRO_ENABLE = yes                                                                
KEY_LOCK_ENABLE = yes 
```

jayliu50, 2017-09-30:
@olivierpons Just an FYI, I believe you are referring to standard macros, and not dynamic macros. This goes for #1775 as well, btw.

fredizzimo, 2017-09-30:
I just submitted a pull request with the fix that I mentioned earlier.

olivierpons, 2017-09-30:
@jayliu50 You're 100% right, not dynamic. I un-plug often my keyboard, do dynamic imply that the macros is lost when you un-plug your keyboard... if it's the case there's no point for me, but if I'm wrong then tell me I'm interested!

fredizzimo, 2017-09-30:
@olivierpons, yes, that's exactly what the [documentation says](https://docs.qmk.fm/dynamic_macros.html)

PR #1777 (2017-09-30): Fix dynamic macros on ChibiOS keyboards
This fixes #1772
closed 2017-09-30 by jackhumbert