qmk/qmk_firmware #1775: Dynamic Macros on Ergodox infinity not repeating

olivierpons, 2017-09-30:
Hi,

All my macros work well but they dont repeat. I have one macro which is Ctrl-Left. If I want to select one word I type Shit+[this macro]
The problem is that it doesnt repeat. If I want to select 4 words I have to type 4 times the key which plays the macro.
Here's the part of the code in my `keymap.c`:
```
const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt)       
{                                                                                
    switch(id) {
        case _CTL: // Ctrl ←
            return MACRODOWN( D(LCTL), T(LEFT), U(LCTL), END);
        case _CTR: // Ctrl →
            return MACRODOWN( D(LCTL), T(RIGHT), U(LCTL), END);
    };
    return MACRO_NONE;
};
```

What should I change to make my macro repeat? In my old Ergodox (not the infinity), with the qmk source the macros were repeating. What am I missing?

fredizzimo, 2017-09-30:
First of all, these are not dynamic macros, but normal macros.

Second, QMK doesn't have, and haven't had any key repeat functionality, it's all handled by the operating system.

You can achieve a repeat of the last key pressed of a macro, by not having the macro send a key up event when you press the macro. In that case you also need to hande the up event yourself, to release the key.

olivierpons, 2017-09-30:
Then I dont know how comes my Ergodox, with the "static" macro "Ctrl-Right" + Shift pressed does repeat, if I keep (key that has the macro assigned) + (Shift) pressed, and the same macro (same syntax on the QMK) on my Infinity Ergodox, doesn't repeat.

Same problem on:

- my big laptop, Windows 7
- my big laptop, Ubuntu
- my small laptop, Mint
- my big PC at home, Windows 10
- my big PC at home, Mint

So I can do macro repeat, it seems it's not related to the operating system.

fredizzimo, 2017-09-30:
Are you sure that you are doing `T(LEFT)` and not `D(LEFT)`? The later will leave the key pressed, so that the operating system can repeat it. But you do need to also handle key up so that you can manually release the held keys. Otherwise they could remain stuck.

olivierpons, 2017-09-30:
I copied/paste from my code and it's working, but not repeating. I'll try to test right now and update my comment here.

closed 2017-09-30 by olivierpons