qmk/qmk_firmware #1789: Ergodox Infinity stuck at QMK logo after flashing

amellors, 2017-10-02:
When I build & program my Ergodox Infinity, the board seems to freeze at the QMK logo screen.  And when I program the bin using dfu-util I get these errors:
dfu-util: invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release

When I program a firmware from the I:C webbased configurer it doesn't show those errors and works as expected.

Any help would be great!

fredizzimo, 2017-10-02:
The dfu warnings are normal. It's a thing that should eventually be fixed, but it doesn't stop the firmware from working. Unfortunately I don't know why it hangs. But here are a few questions to help me figure ot out.

1. Do both halves freeze?
2. Which keymap are you trying to flash? If you are trying a custom one, try the default first.
3. Do you flash it directly from make with `make ergodox_infinity-default-dfu-util`? You can replace `default` with the name of the keymap.
4. Try running `make ergodox_infinity-default-clean` first. What's the full output after you do that and run the above command to flash?

amellors, 2017-10-03:
1) Yes both halves freeze
2) default 
3) I've tried it both ways with default-dfu-util (prefixed with sudo as I'm on linux) & just building using 'make ergodox_infinity-default` and then dfu-util with the .bin in the .builds directory.
4) Clean looks like it succeeds, and then building the full it compiles a bunch of files (It's in a linux VM so hard for me to get the logs off)

amellors, 2017-10-03:
I think the issue was around using a linux VM to try to program the board.

(Got dfu-util.exe working in my windows build environment, was nervous about this because back in April the windows build environment was a complete mess)

Got it all working, closing out this issue

closed 2017-10-03 by amellors