qmk/qmk_firmware #1806: Issue with the default keymap on a Planck

hodyhong, 2017-10-04:
Trying to determine whether this is an hardware issue with my planck. I have got my planck from the latest massdrop. Running it on a mac OSX Sierra. 

With the default mapping, for the alpha layer, 
key for "'" has become ":" and 
keys shift +";:" have become "+*" should be ":""

For the lower layer, 
keys for "yuiop" have become "&'()0", should be "^&*()"
keys for "jkl;:" have become "=~`{}",  should be "_+{}|"

For the raise layer
keys for "jkl;:" have become "-^@[]", should be "-=[]\"

I have refreshed the firmware with default, basic and other keymaps, same issue still occur. 
I have check my solder work they look fine. 
Could this be a firmware issue or hardware issue? Is there anyway I can debug this?

fredizzimo, 2017-10-04:
It sounds like you have a non-English keyboard layout set in the operating system. The layouts in QMK is unfortunately dependent on the active OS layout, and there's no way to work around that.

So if you want to use another OS layout, you have to make your own QMK layout.

hodyhong, 2017-10-04:
I have actually check the system layout. The OS layout is AU English, which is the same as the US layout.

drashna, 2018-03-25:
Sounds like it may not be identical still.

Also, were you able to resolve this?