qmk/qmk_firmware #1814: Dynamic Macros Running Twice

tomatosoupcan, 2017-10-06:
Not sure if I just have something wrong with my config or something, say I start recording dynamic macro 1, then type test, then stop recording. When I run macro 1, instead of typing test, it types testtest. Keymap attached.

Also 

```
uint16_t macro_kc = (keycode == MO(_DYN) ? DYN_REC_STOP : keycode);

    if (!process_record_dynamic_macro(macro_kc, record)) {
        return false;
    }

```
To make the initial layer key be the key to stop recording as well seems to break make.

[keymap.txt](https://github.com/qmk/qmk_firmware/files/1363278/keymap.txt)

tomatosoupcan, 2017-10-06:
Seems like it works fine when the keys to run the macros are not on an OSL() layer

tomatosoupcan, 2017-10-06:
Additionally holding the OSL layer key and pressing the macro key works fine as well, just not as a one-shot I guess.

hiro2016, 2017-10-08:
I might have encountered sometimes similar.    
If I remember correctly, running the macro only on keydown event circumvents it.    
Maybe OSL generates  an extra key release event?

tomatosoupcan, 2017-10-08:
Yeah I was having the same issue on OSL macros a few days ago and corrected my keymap by not letting it return (you can see this in my file), just breaking, so something with the return is probably generating the extra activation.

drashna, 2018-03-25:
Yeah, this is an issue with macros, in general.
Add `clear_oneshot_layer_state(ONESHOT_OTHER_KEY_PRESSED);` before the `return false;`, and I think that may fix the issue.