qmk/qmk_firmware #1815: [Q] How to add a generated file to the build?

hlship, 2017-10-06:
In my fork (https://github.com/hlship/qmk_firmware/tree/cursivetastic/keyboards/ergodox_ez/keymaps/cursivetastic) I have a script that generates a `gen-keymap.h` file that is included from my largely-skeleton `keymap.c` file.  

However, especially with the recent changes to the build system, I have been unable to figure out how to add a rule, somewhere, to generate the `gen-keymap.h` file. Is this kind of thing even supported?  If not, can it be added?

My rules are very simple:

```
gen-keymap.h: src/cursivetastic/keymapping.cljc
	lumo src/cursivetastic/keymapping.cljc

keymap.c: gen-keymap.h
```

... but I don't see me submitting a PR back to qmk_firmware because `cursivetastic` is very, very, very, (very!) specific to my needs.  But I continue to track qmk_firmware as my upstream, and want a process where I can just execute a quick make when I change my `keymapping.cljc` script.

drashna, 2018-03-25:
I don't think there is a good answer for you. 

You'd want to rebase (`git rebase upstream master`, or `git pull --rebase`).  And then you'd need to change things.

And I don't think that your `makefile` will work at all.  And you'd probably need to modify `/tmk_core/build_avr.mk` to re-add this

drashna, 2018-10-21:
Were you able to resolve this?

hlship, 2018-10-22:
Nope!  Haven't had the spare cycles to figure out a better solution. I just have an ugly manual process.

drashna, 2018-10-22:
Okay, totally understand that.

closed 2019-12-21 by stale[bot]