qmk/qmk_firmware #1819: on 2nd fast tap hold action never valid

FraMecca, 2017-10-07:
Hi,
I have a ergodox with a mod key that is: 
LT(SymbolsLayer, KC_SPACE)

When I press the mod key one time and then hold it, the symbol layer is never pushed and I can't use the hold action.

Is there an option to disable this feature?

hiro2016, 2017-10-08:
That is a feature related to tap dance action.
To disable it, uncomment the line `//#define NO_ACTION_TAPPING` in config.h.  
There are several config.h files and rather confusing but editing `keyboards/ergodox/ez/config.h` did the trick fro me.   
  
But if your problem is:  
 - not being able to use LT after typing spaces  
Maybe tapping term or other variables are not set right. I have no trouble using LT after typing space key.

 - have to use LT often and consecutively to input math symbols, phonetic alphabets, etc
Then LT might not be the real solution for you. LT requires you to press down a key for some period of time and is fairly uncomfortable.   

For the latter case, there is no single solution that suits all cases.  
But you can write a macro or custom action code that suits your particular needs like I did:  
https://github.com/hiro2016/ergodox_gui_configurator/blob/master/README_CLT.md    
Well, the code is not a good sample and ugly but it should provide a preview of just how much freedom tmk is letting us have.

FraMecca, 2017-10-20:
If I enable NO_ACTION_TAPPING I can't use the LT macro anymore, and the key act as if it is always a space.

hiro2016, 2017-10-21:
Real sorry looks like I did not test with LT when I tested how things go with NO_ACTION_TAPPING enabled.  
  
There could be a real solution to your problem, but that I don't know.    

Either write your own custom action code   
Or  
Use a buggy code I'm using.    

As to custom action code, something like this might do(not tested):  

```

enum custom_keycodes {
  LT_ONE_OR_ENTER = SAFE_RANGE
};
uint16_t custom_lt_timer;
keypos_t prv_key_pressed;
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
#define IS_KEYPOS_SAME(keyone,keytwo)  ((keyone.col==keytwo.col)&&(keyone.row==keytwo.row))
#define ANOTHER_KEY_PRESSED (!IS_KEYPOS_SAME(prv_key_pressed, record->event.key))
       prv_key_pressed = record->event.key;
	if(keycode == LT_ONE_OR_ENTER) {
		if(record->event.pressed) {
			custom_lt_timer = timer_read();
			layer_on(1);
		}else{
			layer_off(1);
			if (timer_elapsed(custom_lt_timer)<200 && (!ANOTHER_KEY_PRESSED)) {
                            register_code(KC_ENT);  
                            unregister_code(KC_ENT);
			}
		}
	}
	return true;
}
  
```

FraMecca, 2017-10-21:
I have added to config.h:

```
#define IGNORE_MOD_TAP_INTERRUPT 
#define TAPPING_TERM 150
```
and it seems that a low tapping term did the trick.

Thank you for the help

closed 2017-10-21 by FraMecca

hiro2016, 2017-10-23:
Glad you have solved it yourself.  
Sorry could not help much and thanks for sharing that info with me.