qmk/qmk_firmware #1844: Nyquist stuck as Left Hand Master

LaikeSF, 2017-10-15:
No matter what I do, the Nyquist will only work as the left hand master. I am trying to switch to EE_Hands. I have flashed each side with the EEPROM of each side from the Nyquist folder. I'm not entirely sure what else I may be doing wrong. My current config is as follows:

```
#ifndef CONFIG_USER_H
#define CONFIG_USER_H

#include "config_common.h"

  #define USE_I2C

 #define EE_HANDS

#undef RGBLED_NUM
#define RGBLIGHT_ANIMATIONS
#define RGBLED_NUM 12
#define RGBLIGHT_HUE_STEP 8
#define RGBLIGHT_SAT_STEP 8
#define RGBLIGHT_VAL_STEP 8

#endif
```

drashna, 2017-11-29:
Have you tried reflashing the EEPROM files?

LaikeSF, 2017-11-29:
Yes, I've done that several times without any effect.

drashna, 2018-03-25:
You're using RGB, right? 
#2399 and #1576.

Update the repo, then retry.  Or disable RGB.

LaikeSF, 2018-03-26:
I've pulled the latest build from the repo and tried again. I've also tried commenting out all the RGB related parts in the config. Same problems.