qmk/qmk_firmware #1845: dead link in the QMK manual

morte-rictusgrin, 2017-10-15:
The link to [default Clueboard keymap](https://github.com/qmk/qmk_firmware/blob/master/keyboards/clueboard/keymaps/default/keymap.c) leads on the [Keymap Overview](https://docs.qmk.fm/keymap.html) leads nowhere. Most likely, cause clueboard has been forked to clueboard_17, clueboard_66 and clueboard_60. The entire reference is now useless: keymap for clueboard_66 is way different than described in the manual.

fredizzimo, 2017-11-19:
This was fixed by #1845

closed 2017-11-19 by fredizzimo

morte-rictusgrin, 2017-11-19:
still no: https://github.com/qmk/qmk_firmware/blob/master/keyboards/**clueboard_66**/keymaps/default/keymap.c leads nowhere again. Change underscore with slash and you're good.

https://github.com/qmk/qmk_firmware/blob/master/keyboards/**clueboard/66/**keymaps/default/keymap.c

mmikeww, 2017-12-01:
I have fixed this in PR #2083