qmk/qmk_firmware #1860: XD60 with iso and split backsapce

Gertsch, 2017-10-18:
Hi guys,
I bought a xd60 set and built all the physical stuff.
Now I'm a little bit stuck at programming this thing^^
Here is the [layout](https://goo.gl/SResHX).
I'm not really sure how I can translate my layout to QMK.
Was trying https://kbfirmware.com/ to compile my layout, but I'm not sure how many col/rows I need to set.
And how pins need to be routed.

jceaser, 2017-10-18:
First of all, I'm sorry, I don't know how to use the fancy GUI tools:

if your board in is not in the list of supported keyboards you will have to either submit a change to the kbfirmware.com project and wait for it to show up or just use QMK directly. Your keyboard is supported, I would do the following:

1. `git clone https://github.com/qmk/qmk_firmware.git`
2. then copy the directory `keyboards/xd60/keymaps/default` naming it anything you want
3. change keymap.c

Now your ready to compile. I would suggest you use https://www.docker.com. Download and install this, when done run the following:

1. `docker build -t qmk `
2. `docker run -e keyboard=xd60 -e subproject= -e keymap=<your_layout_name> -v <full_path_to_project>:/qmk/ qmk`

change your_layout_name to what you named the directory you copied up above, and full_path_to_project to where you downloaded qmk to. That should compile the project. Once done, you upload with dfu_util. as such `dfu-util -D .build/x60<some_file>`. I think there is a way to get docker to also do the install, but I forget how.

In my Dockerfile I have `#CMD echo making ${master}  ; make ${keyboard}-${keymap}-dfu-util` which I think will do everything for you, but I have it commented out. For one of my more funny boards I do it the other way and call dfu-util by hand because it is strange, maybe you can try it this other way and see how you go.

drashna, 2018-03-25:
This keyboard was added, so you should just need to copy and edit the default keymap now. 

https://docs.qmk.fm/newbs_building_firmware.html

I'm closing this now.  If needed, reopen or open a new issue.

closed 2018-03-25 by drashna