qmk/qmk_firmware #1864: Ergodox infinity only works with right hand master

ckaros, 2017-10-18:
I have two ergodox infinities flashed with the same qmk firmware. One works perfectly, while the other only really works with the right hand as master. If I plug in the left hand as master, the keyboard works fine for about 5 seconds, then the key leds on the right hand turn on and that side stops working. The left side continues to work, the but the right hand is dead. If I flash with kiibohd firmware, it works fine with left or right as master. Not sure what is happening here, maybe someone has an idea?

Before anyone asks, yes I am flashing the left with MASTER=left and the right with MASTER=right.

DanielGGordon, 2017-10-19:
I had the same issue - for me it was the USB cable. The infinity is really picky. There is a Cable Matters USB 3.1 type A to USB C that worked for me. It must be USB 3.0 or higher.

ckaros, 2017-10-25:
I got a new usb cable. The problem is much better. The right have does work now, but it takes much longer to stop working. Additionally, if I tilt/jostle the keyboard enough, it reboots. This it, the back light fades out, the screen says suspending, and the keyboard restarts, but then fails to work. 

This definitely sounds like a power issue. The strangest part is that the board works perfectly fine with kiibohd firmware. Is there a parameter in the code the controls the power delivery to the slave board?

Trevortds, 2018-01-31:
I am also seeing this problem. Slave board would suddenly stop working, display the QMK logo, then "SUSPENDING". Switching the right board to master seems to have solved this problem, at least in the short term.

drashna, 2018-03-25:
Is this still happening? 
(eg, okay if I go ahead and close this)?

Trevortds, 2018-03-25:
My issue turned out to be an ESD problem with my powered usb hub. The problem persisted when I plugged into right hand master, but went away when I started plugging directly into my laptop. 

So I think it's safe to close this. For googlers who come in later: check cables and power supply, power screwyness can cause these symptoms.

drashna, 2018-03-25:
Glad to hear it wasn't your keyboard, at least! 
And yeah, try everything between computer and keyboard!

closed 2018-03-25 by drashna