qmk/qmk_firmware #1884: Incorrect keycodes sent in extended+localised space cadet

Krakob, 2017-10-22:
[Here's](https://gist.github.com/Krakob/771b08876a20acf0b142e35acd34755c) a slightly shortened version of my current firmware for the DZ60. I based it on [this help](https://github.com/qmk/qmk_firmware/issues/1404#issuecomment-312422434) I received back in July

I'm trying to replicate space cadet shift as a generalised function that can be used for all types of brackets. On the swedish layout, the keys are 

| Character | Key combination | My cadet mapping
| --- | --- | --- | 
| { | AltGr + 7 | Tap left GUI
| } | AltGr + 0 | Tap right GUI
| ( | Shift + 8 | Tap left Shift
| ) | Shift + 9 | Tap right Shift
| [ | AltGr + 8 | Tap left Alt
| ] | AltGr + 9 | Tap right Alt

< and > are on their own key, I haven't integrated them into this.

The results of this are strange, to say the least. I'm using [screenkey](https://github.com/wavexx/screenkey) to view sent inputs, this is what I get

| Key | Sent on tap | Sent on hold + a |
| --- | --- | --- |
| Left shift | Shift + 8 | Shift + A |
| Right shift | Ctrl + Shift + 9 | Ctrl + Shift + A |
| Left GUI | Ctrl + Alt + 7 | Super + a |
| Right GUI | Ctrl + Alt + 0 | Ctrl + Super + a |
| Left Alt | Ctrl + Alt + 8 | Alt + a |
| Right Alt | Ctrl + Alt + 9 | Ctrl + Alt + a |

* This means the only key working as intended is left shift.
* Alt keys that are supposed to send AltGr instead send Ctrl+Alt.
* All lefthand keys work as intended when held, while all right hand keys add Ctrl to their intended key, although I can't tell if Right Alt is sending Left Alt + Left Ctrl or Left Alt + Right Ctrl.

The only thing I can get out of this all is that MOD_RALT actually sends MOD_LALT and MOD_LCTL, or perhaps MOD_RCTL. This works on Windows in order to send {[]}, but not on Linux. Other than that, I am absolutely clueless as to what's going wrong here.

Krakob, 2018-01-06:
I'm offering a €30 bounty if someone would implement this natively in qmk, that is:

* User is given a simple way to define a generalised space cadet key in the form (or similar to): `#define my_pseudokeycode cadet(key_on_hold, mod_on_tap, key_on_tap)`. Example usage `#define LCTL cadet(KC_LCTL, MOD_RALT, KC_7)`.
* Function must support AltGr (**not** the same as ctrl + alt)!
* User should not have to implement an `action_function` or similar.

Bounty can be paid via PayPal (international) or Swish (Sweden only) on the conditions a) the patch is merged into QMK and b) I can verify its functionality.

jackhumbert, 2018-01-06:
If you use `register_code` instead of `register_mods`, while swapping in the equivalent keycodes in for the mods (`KC_LCTL` instead of `MOD_LCTL`, etc), do you get the same result? Right-hand mods aren't ideally implemented in the internal mod system at this point, but should work on raw keycode presses/sends.

PureSpider, 2018-01-06:
AltGr on layouts that support it is simply RAlt on all the others. Try using RAlt instead of AltGr?

Krakob, 2018-01-06:
@jackhumbert -- I will try it tomorrow, thanks for the tip!
@PureSpider -- AltGr and RAlt are synonymous in terms of keycodes, it's just semantics, kind of like return vs enter. Unfortunately, implementation might not be correct, as seen in my issue.

Krakob, 2018-01-09:
@jackhumbert -- after my liberal interpretation of 'tomorrow', I've gotten it working! Thank you!

It's late o' clock so I'll get around to linking the solution (currently in a private gitlab repo), closing, and moving the bounty to a separate issue tomorrow. I'll still uphold it since this method was pretty cumbersome anyhow and QMK is an awesome project that I'm happy to support :)

Krakob, 2018-01-10:
Solution: https://gitlab.com/Krakob/qmk_firmware/blob/krakob-xd60/keyboards/xd60/keymaps/krakob/keymap.c

Includes a solution for hold-layers instead of mods, as well since I only use one GUI key and had to put in an opposite for it.

Again, thanks a bunch for the help! <3

closed 2018-01-10 by Krakob