qmk/qmk_firmware #1902: How to fix ergodox infinity backlight breating

Twwow, 2017-10-25:
When I try to compile ergodox infinity with backlight breating
function screen shown error as below.
****************************************************************************************************
Linking: .build/ergodox_infinity_default.elf                                                        [ERRORS]
 |
 | .build/obj_ergodox_infinity_default/keyboards/ergodox_infinity/keymaps/default/keymap.o: In function `process_record_user':
 | X:\qmk_firmware/keyboards/ergodox_infinity/keymaps/default/keymap.c:194: undefined reference to `breathing_enable'
 | collect2.exe: error: ld returned 1 exit status
 |
make[1]: *** [tmk_core/rules.mk:287: .build/ergodox_infinity_default.elf] Error 1
Make finished with errors
make: *** [Makefile:533: ergodox_infinity:default] Error 1
****************************************************************************************************
I apologize for not being word-perfect in English.

drashna, 2017-10-25:
Well, it looks like you're calling "breathing_enabled", which isn't actually defined in your code (on line 194, of your keymap file). 

Make sure that you have a "config.h" file, with `#define BACKLIGHT_BREATHING` in it somewhere, and then try recompiling.

Twwow, 2017-10-26:
@drashna thanks for your reply.
#define BACKLIGHT_BREATHING already put in "config.h" file.
On my Planck keyboard backlight breathing is working well.
Compling error only occur on ergodox infinity keyboard.

drashna, 2017-10-26:
Okay then.  Have you tried `make ergodox_infinity:clean`, and then recompiling?  And maybe add "BACKLIGHT_ENABLE=yes" to the make command, *just* in case? 

Aside from that, not much else I can help with here. Sorry.

skipperwarlock, 2018-02-04:
@Twwow were you ever able to fix this? Having the same issue

drashna, 2018-03-25:
I've since learned that most of this stuff is handled by the visualizer.c file.  Try messing with that, and seeing if somebody else's code will work for you.

fredizzimo, 2018-04-07:
Backlight breathing is not supported on the ARM keyboards. But it is possible to archieve a similar effecty by creating your own custom visualizer, and an animation that does the brething. 

That process is unfortunately not documented but take a look a `visualizer.c` and `simple_visualizer.h` in the Ergodox Infinity directory for some examples. You will need to make a similar visualizer in a file `visuzlizer.c` in your keymap directory.