qmk/qmk_firmware #1907: Question: How I activate layer and modifier in a single key?

zer09, 2017-10-26:
Hello good day.
I been trying to make that when I hold a single key it will activate a momentary layer and also the shift key, but without success. I tried mixing some ```SFT``` keycodes with the ```MO``` but cant make it work.

Please help, thanks.

drashna, 2017-10-26:
I'm not sure of the "proper" way to do this, but this is how I'd do it: 

```
void matrix_scan_user(void)
{  // runs frequently to update info
    uint8_t layer = biton32(layer_state);
    static uint8_t current_layer = 0;
    static bool has_layer_changed = false;
    static bool is_shifted = false;

    if (layer != current_layer)
    {
        has_layer_changed = true;
        current_layer = layer;
    }
    // Check layer, and apply color if its changed since last check
    if (has_layer_changed)
    {
        if (layer == _MOLAYER)
        {
            register_code(KC_LSFT);
            is_shifted = true;
        }
        else
        {
            if (is_shifted)
            {
                unregister_code(KC_LSFT);
                is_shifted = false;
            }
        }

        has_layer_changed = false;
    }
 };
```

Where `_MOLAYER` was the layer in question. 

This would "register" (press) the "shift" press when switching to this layer, and unregister (unpress) it when leaving the layer (and ONLY when leaving this layer).

zer09, 2017-10-27:
Thanks.

If I understand it correctly, it will always register the shift key as long as you switch to that layer?

Because my plan is, the layer can be activated via ```MO``` and ```TO``` and will only register the shift key if it was activated with ```MO```. But in your code I think we can add another condition to identify if it is activated by ```MO``` or ```TO```. Is there any flag for that?

drashna, 2017-10-27:
Yes, it would. 

If that's not what you want, then instead, create a custom keycode for both.  Set "bool is_shifted = false;" at the top, and set it in the keycode.

It would be a bit more difficult/complex here.  Basically, adding another variable in the keycode.

zer09, 2017-10-27:
Thanks for the hint. I really need to explore it more.

Is it ok to keep this issue open? maybe others have their own implementations to share.

zer09, 2017-10-29:
@drashna BTW last question is there way to identify which column/row has been pressed and or how the layer  toggled (I mean identify if it is toggled by the ```MO``` or ```TO``` etc.)?

Thanks.

drashna, 2017-10-29:
As for keeping the issue open, until it's resolved to your satisfaction, it should be "okay". But I'm not one of the maintainers. 

As for how it was switched, I don't think so.  However, again, custom keycode, and define a boolean. Set it based on what you want/need to do.

closed 2017-11-03 by zer09