qmk/qmk_firmware #1923: [Request] improved handling for chorded MO layer toggling

jsmith8, 2017-10-29:
In a 4-layer setup with MO(1) and MO(2) mapping on keys (e.g.,) A and B of layer 0, MO(3) on layer 1's B, and MO(3) on layer 2's A, chording the Momentary layer toggles doesn't work quite as I would hope.

The state stays in layer 3 when the first MO trigger held is released before the second.
I.e., +A (L1) +B (L3) -A (still L3, but would prefer L2) -B (L0)
or: +B +A -B (still L3 but would prefer L1) -A.

Is there a known better usage pattern to achieve what I'm looking for, or would a code change be required for this?

drashna, 2018-03-25:
Chording requires that you hold the keys down, right? 

If so .... anytime you switch layers, the keys get "cleared" (unregistered/unpressed).... so .... swapping layers in the middle of a chord won't work.

fredizzimo, 2018-04-07:
Yes, `MO` currently works like `INGORE_MOD_TAP_INTERRUPT` is always enabled, which means that they are processed as taps as soon as another key is pressed, unless you hold it for longer than `TAPPING_TERM` first. This will be changed to support the same type of configurability as the `MT` macro.

So for now, the only option is to make `TAPPING_TERM` smaller.

jackhumbert, 2020-03-17:
@jsmith8 does `TAPPING_TERM` work for you, or did you figure out another way to address this?

zvecr, 2020-10-20:
Closing as there hasn't been any movement on this, we can reopen if its still an issue.

closed 2020-10-20 by zvecr