qmk/qmk_firmware #1938: some keys don't work while recording a dynamic macro

mpjuers, 2017-10-31:
If I begin recording a dynamic macro, the 'a' key does not work for as long as I'm recording. As soon as I stop recording, the issue goes away. I'm on a zeal60 and I wasn't sure where to create the issue (here or on Wilba's fork) since I don't know if the issue affects both. Perhaps someone can replicate this on vanilla qmk firmware? [Here](https://github.com/mpjuers/qmk_firmware/tree/zeal60/keyboards/zeal60/keymaps/mpjuers) is my keymap directory.

drashna, 2018-03-25:
Is this still an issue? 

And have you tried updating your repo (if cloned with git, then git rebase or git merge)?

drashna, 2018-10-22:
Did you get this resolved?

mpjuers, 2018-11-23:
Sorry, I've been buried in email and didn't see the update for this. I will reflash it some time over break. In addition to the problem I reported in the original issue, when typing in all caps (while holding shift; I have no caps lock so I haven't tested this in that case), the same key, A breaks the shift and I must release it and press it again. So the string "AFAIK" becomes "Afaik". I suspect the issues are related because they affect the same key. I will keep you posted.

jackhumbert, 2020-03-17:
Sorry to ask the same thing again, but we recently merged the dynamic macro stuff into core - has the behavior around this changed at all?

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc