qmk/qmk_firmware #1942: Support for Massdrop x Input Club K-Type

KMontag42, 2017-10-31:
Wuddup yooooo!

Any plans to add support for the new K-Type to QMK??

belak, 2017-11-02:
Adding basic support wouldn't be hard, but adding full LED support will most likely be a challenge. Last I knew there was still some work being done on the visualization front.

If you're interested in doing the port yourself, it should be as simple as copying a different TKL keyboard and updating it with the correct info (CPU, matrix layout, etc) from the following links:

* https://github.com/kiibohd/controller/blob/master/Keyboards/k-type.bash
* https://github.com/kiibohd/controller/blob/master/Scan/K-Type/matrix.h
* https://github.com/kiibohd/controller/blob/master/Scan/K-Type/scancode_map.p3.kll (Technically it should just be scancode_map.kll, but it's a symlink so this link is more useful at the moment)

belak, 2017-11-03:
I started on it but haven't gotten it running yet. I've got most of the busywork and scancodes done... but I haven't been able to find any info on which chip this is built on (among other things) so it's definitely missing some important settings.

KMontag42, 2017-11-03:
If you have a branch I'd be happy to help out however possible, let me know.

<3

belak, 2017-11-03:
Once I have a build that flashes properly, I'll look at opening a PR. There is a branch at https://github.com/qmk/qmk_firmware/compare/master...belak:k-type-support but keep in mind this doesn't work at all right now.

belak, 2017-11-03:
This patch is needed to get the firmware to boot (we'll look at including this in a board file similar to the clueboard support eventually). However, USB support is still broken.

The following code comes from here:

* https://github.com/kiibohd/controller/blob/1d8ee5786d80421eacb07d68b9133fbb3a31f53a/Lib/mk20dx.h#L728 
* https://github.com/kiibohd/controller/blob/1d8ee5786d80421eacb07d68b9133fbb3a31f53a/Lib/mk20dx.c#L530

```
chibios-contrib % git diff
diff --git a/os/hal/ports/KINETIS/K20x/hal_lld.c b/os/hal/ports/KINETIS/K20x/hal_lld.c
index e6eeed8..cb8edd8 100644
--- a/os/hal/ports/KINETIS/K20x/hal_lld.c
+++ b/os/hal/ports/KINETIS/K20x/hal_lld.c
@@ -93,8 +93,10 @@ void hal_lld_init(void) {
  *
  * @special
  */
+#define WDOG_TMROUTL            *(volatile uint16_t *)0x40052012^M
 void k20x_clock_init(void) {
 #if !KINETIS_NO_INIT
+  while (WDOG_TMROUTL < 2); // Must wait for WDOG timer if already running, before jumping^M

   /* Disable the watchdog */
   WDOG->UNLOCK = 0xC520;
```

KMontag42, 2017-12-06:
sorry for the late reply, I will be trying this out over the weekend!!!!

PR #1963 (2017-12-14): Initial K-Type support
Fixes #1942

In order to run this, you need my fork of ChibiOS-Contrib (The fix has been merged upstream, but QMK needs to be updated to the latest versions of ChibiOS and ChibiOS-Contrib. My fork has the needed patch applied to the latest commit used by QMK). Also note that only the left USB port currently works properly.

Instructions:

```
# From the root of the qmk repo, make sure the fork is added
# as a remote and check out the k-type branch
git remote add belak https://github.com/belak/qmk_firmware
git fetch --all
git checkout k-type-support

# Switch to belak's c [... cut]
closed 2017-12-14 by jackhumbert