qmk/qmk_firmware #1950: Cannot do crouch jump (c + space button combination) in PUBG, only seems to affect QMK-based keyboards

ccristob, 2017-11-01:
I seem to have a problem doing the crouch jump in PUBG but only have an issue when I'm using keyboards on QMK.

For those that don't know, a crouch jump in PUBG is done by pressing crouch and jump at pretty much the same time. It's done to get to higher places, but most importantly to jump out of windows. I can get it consistently on other boards. (YMD96, Zhuque, even the K-Type). 

However, once I use a QMK keyboard (KBD75, DZ60, XD60), it pretty much becomes impossible. This is with the same switches on both type of boards. (Kailh Purple)

Anybody know what's up? Could it be the polling rate? Not quite sure on how to change that on QMK as all I can see in the code are references to frequency values. Hopefully someone has a solution out there. 

The combination is C and SPACE at the same time. (that's to jump straight up) Shift + C + Space is needed for jumping out of windows. However, just jumping straight up is not working so there's definitely issues with just C + SPACE

I've actually already tried using a macro (using sendstring) for a single key to accomplish this and I'm getting the same results. The macro works fine outside of the game and I get "c " if I press it in notepad for example. The same macro when configured on a mouse button using the Logitech gaming software works flawlessly, so something looks to be up. Both macros in QMK and Logitech are set to have no delays.

Maartenwut, 2017-11-06:
I can jump trough windows without a problem on my Tada68 running QMK. Which keyboard/layout do you use?

ccristob, 2017-11-06:
KBD75, DZ60, XD60. Standard ANSI layout, really. Consistently can't do it
on these keyboards, and can consistently do it on regular mech keyboards
from OEMs.

seebs, 2017-11-22:
QMK, by default, restricts key events to one event per scan. I have a patch in master that allows increasing this. This doesn't directly affect polling rate, exactly, but affects the rate at which key events will show up (with a couple-few milliseconds of delay for multiple keys in some cases).

Grab the master tree, and build with QMK_KEYS_PER_SCAN set to, say, 4 or so, see if that helps?

MarioRicalde, 2018-02-12:
Did this actually help? What's the status with this issue? Seems abandoned.

ccristob, 2018-02-12:
Unfortunately I'm not able to test this now as PUBG has removed the crouch+jump from their game and even if it's there, it's also not useful anymore now that the game has vaulting. Much thanks to seebs though, as what he suggested is most likely what the issue was here.

MarioRicalde, 2018-02-12:
Awesome. You can close this issue! 👍

closed 2018-02-12 by ccristob