qmk/qmk_firmware #1958: way way too high default debounce on ergodox ez

nathanejohnson, 2017-11-03:
I loaned my ergodox out to a coworker and just got it back today, and I needed to reflash back to my layout.  When I did, I had issues typing many key sequences (my password for instance), any time where I'd be typing the same key twice in a row, and when I dug in I found this PR was biting me:

https://github.com/qmk/qmk_firmware/pull/1279

I think the general principal of debouncing presses of the same key is a neat idea, but changing from 5 -> 15 was way way way too much in my case.  I reverted back to 5 and the issue disappeared.  I'm using Cherry switches with an original hand-built ergodox (pre-Infinity Massdrop kit).  Two of my coworkers just purchased EZs, and one uses the same map as I do, so I'll try it on theirs and see if it seems different on their hardware.  But at the very least, this should probably have a comment somewhere.

jackhumbert, 2017-11-03:
I think the EZ was having some chatter/debounce issues (which have now been resolved I believe), and that code helped to alleviate the problem, but @ezuk might have more input on that!

In any case, lower it as much as you can while not affecting chatter :)

drashna, 2017-11-03:
Well, worst case, you can undefine and redefine it the debounce value in your keymap's config.h file.

nathanejohnson, 2017-11-03:
@drashna, sure, there are many ways I can use my own value of the DEBOUNCE, but I thought opening an issue might be useful just to continue the discussion around the topic since the original PR has been closed and merged.  My personal opinion is that 15 is not a very good default, but I'm just one person and not a very important one at that ;)

drashna, 2017-11-03:
Actually, here's a good question: 

https://github.com/qmk/qmk_firmware/blob/master/keyboards/ergodox_ez/matrix.c#L53

So, if DEBOUNCE wasn't defined as 15 in the keyboard's config.h, it would be defined as 5, by the keyboard.

Though, from what I understand here, that lower than 15 may cause issues, and probably due to the hardware (the communication with the IO expander chip being the issue?). 

Aside from that, I've been running it as "5" since posting, and haven't noticed an appreciable difference in behavior.

ezuk, 2017-11-03:
Monitoring this. 15 seems to be working well for the vast majority of users, but if I get more feedback we may drop back down to 5.

nathanejohnson, 2017-11-03:
@ezuk thanks for looking into it!

for me, I can reproduce it doing the following.

with DEBOUNCE 5
I feel free, freely freeing free things.

with DEBOUNCE 15
I fele fre,e freley freieng fre things.

nathanejohnson, 2017-11-03:
@ezuk I will test with one of my coworkers' EZ today to see if it's maybe a hardware difference.  I have another 'original' ergodox that is still running the pre-PR debounce code, that one set to 5 cycles, and I never noticed an issue with that one either (both of mine are stiffly sprung, one being cherry mx clear and the other being mx green, not sure if that makes a difference wrt required debounce times).  I am a reasonably fast typist so that may be part of reason I notice the issue while others don't.  Also, for what it's worth, I actually lowered my DEBOUNCE to 3 on this keyboard and so far I notice no unwanted keypresses, and anecdotally it "feels" more responsive to me than 5.

nathanejohnson, 2017-11-03:
@ezuk so plot twist, I can't reproduce this on my work ergodox with 15 scans.  I'll do more testing this weekend when I have several keyboards all in one place and see if I can figure out what the hell the difference is.

drashna, 2017-11-03:
Well, I don't think it's the switches.  IIRC, the actuation curve for clears vs greens is somewhat similar.  

My guess would be a bad TRRS cable maybe (or bad connection/dirt/lint), maybe a bad solder joint, or issue with the controllers.  (but that's just a guess/shot in the dark).

nathanejohnson, 2017-11-04:
@drashna I would be suspicious of a faulty connection issue if I had to take the DEBOUNCE value up really high to make the issues go away, or if no setting for DEBOUNCE made the thing behave.  Also this keyboard has been solid and in regular use for over two years now with no issues previously.  In this case, the only issue I'm having when the DEBOUNCE is set to 15 on one specific keyboard, then it eats keypresses (and only consecutive presses of the same key at that).  It's almost as though the clock on one of the teensy's is a lot slower than the other, but that shouldn't be possible afaik.  They're both the same model according to the Teensy programmer, the odd one source from a Massdrop drop, the other one sourced from mechanicalkeyboards.com.  Anyway, maybe this one ergodox is a bit of a unicorn.  With a DEBOUNCE setting of 3, both of my keyboards hum along wonderfully with no ghosted keypresses and no swallowed ones either.

kdknowlton, 2017-11-08:
Just in the name of further feedback, I believe I am also having problems with the higher debounce setting. I've been using the UI at configure.ergodox-ez.com, and my previous firmware from several months ago gave me no problems. I generated a new on yesterday and have consistently had problems with a second press of the same key not registering or registering late in normal typing. 

Since I'm using the UI I can't change the debounce setting or be sure what it is, but I'm assuming it uses the default of 15 and that it was 5 when I generated the old firmware.

The actual hardware I'm using is marked as batch 1b by Tibbo Technology Inc. and has Gateron Brown switches.

ezuk, 2017-11-09:
Thanks, that's good feedback @kdknowlton - I am working on exposing debounce settings (and other `config.h`) options via the configurator, hopefully in the next few weeks. Are you currently stuck with customizing your keyboard? You can ping me at contact@ergodox-ez.com also.

kdknowlton, 2017-11-09:
Actually I'd been meaning to get set up to build keymaps locally and this finally pushed me over the edge. I got the source from the configurator and flashed a new version using debounce 5 this morning. It's working great so far.

Thanks for the offer, and thanks again for all the work you do to make the Ergodox available and easy to work with.

ezuk, 2017-11-09:
Hey, that's awesome!! Very happy to hear you're building locally and this is resolved. :) I'm still working to expose those options, but IMHO the more people build locally, the better -- full control + open source. :)

pjroth, 2018-03-21:
I am having this problem where I'm seeing dropped key presses when rapidly pressing the same key (it just happened when typing the double "p" in dropped).  Strangely it seems like it suddenly got worse today (I rarely noticed this until today) and not related to me re-flashing recently since I've been using this layout/config/build for more than 2 weeks.  

I've tried rebooting the keyboard (removing USB momentarily) and that has no impact.

I'll try re-flashing and if nothing else works I'll figure out how to build myself and adjust the DEBOUNCE value as suggested here and report back.

My config:  https://configure.ergodox-ez.com/keyboard_layouts/qvvaav/edit and I'm using a linux Ubuntu 16.04.4 LTS.

Note the best test I have to reproduce this is when typing ellipses, I never get three periods but after three quick key presses I get 2 periods.

pjroth, 2018-03-21:
I built off master just now using default ergodox-ez and default keymap and flashed this version to my ergodox-ez. This ELIMINATED my problem of missing the second quickly typed key.  Typing is also noticeably snappier.  I'm guessing @ezuk your https://configure.ergodox-ez.com tool is using an older version than this latest master revision I used and the difference between those versions is causing this issue?  But of course I'm just guessing.

Here is the rev I used:
```
$ git log -1
commit f0932a8716dc946322c5ebae7f75eaa275c6220c
Author: Drashna Jaelre <drashna@live.com>
Date:   Wed Mar 21 00:09:51 2018 -0700

    Update info.json for Ergodox EZ (#2577)

```

I used this command to build this:
`docker run -e keyboard=ergodox_ez --rm -v $('pwd'):/qmk:rw edasque/qmk_firmware`

Note I also tried v1.3.3 of default config from https://configure.ergodox-ez.com/keyboard_layouts/new just now and that has same bad behavior as my custom keymap.  Missing double tapped keys.

drashna, 2018-03-21:
I suspect that it's the `QMK_KEYS_PER_SCAN` option that may be the issue. 

Try adding `#define QMK_KEYS_PER_SCAN 4` to your config.h, and see if that helps.  And try playing with that value (set it to 1, and 8).  

Otherwise, you can see a history of commits to just the ergodox board here: https://github.com/qmk/qmk_firmware/commits/master/keyboards/ergodox_ez

pjroth, 2018-03-22:
Thanks for the advice @drashna however using the value of 15 from master my problem is no longer present after I built the code from scratch myself (as compared to using https://configure.ergodox-ez.com where the hex file compiled for me is causing my keyboard to miss quick taps).  This issue is maybe not the correct place for this issue, it seems like a bug with https://configure.ergodox-ez.com to me but it's all a little strange so I can't be sure.  I didn't flash anything and this problem just started to happen which is very odd to me. 

TL;DR Using https://configure.ergodox-ez.com hex files cause quick taps to miss keystrokes but compiling locally myself from master (f0932a87) works great and no missed keystrokes....

drashna, 2018-03-22:
Okay, glad to hear it.

I think there was some optimization that Seebs did a while ago that just hasn't been added to the Ergodox EZ Configurator yet. And namely, I think the "QMK_KEYS_PER_SCAN" code hasn't been added, yet.

ezuk, 2018-03-22:
That must be it! Thank you @drashna -- I'll be upgrading the qmk version on the configurator over the next few days and will check back in here to confirm, and we can test it then.

ezuk, 2018-03-23:
@drashna @pjroth I've upgraded the qmk version running on the configurator. @pjroth  - Would you like to maybe check it out and tell me if that resolves the issue for you?

pg30123, 2018-03-23:
It seems to work for me.

I was also having the same issue that repeated key strokes are constantly getting ignored. After seeing this update I went to the configurator and started with a clean copy of 1.3.4 default layout,  changed the keymap to my liking, and flashed it to the keyboard, and it worked. :D

pjroth, 2018-03-23:
Thanks so much @ezuk for making a change so quickly, much appreciated!

So I cloned my previous config and built and downloaded a new hex file using this config: 
 https://configure.ergodox-ez.com/keyboard_layouts/kjbmrv/edit

I loaded this via teensy loader on linux and my keyboard has NO repeated keystrokes.  That's good, but....

Next I said, lets reproduce the previous issue by loading the old problematic version that I have saved on my PC.  So I loaded my saved hex file that was causing me problems two days ago when I first commented on this issue.  And there are NO problems with repeated keystrokes..... :(  I tried a few loads, power cycling the keyboard, etc ,etc and I now am unable to reproduce the problems from 2 days ago.  As a reminder I had not loaded a new hex file when I initially observed this problem.  My keyboard just randomly started exhibiting this problem one day when I came into work (where I keep my keyboard).  

Seems like something more intermittent is going on.  I'll keep a close watch on things and see if I can learn anything that might be useful.

drashna, 2018-03-23:
If that's the case, maybe it's a bad cable (TRRS, or USB) then, maybe?

ezuk, 2018-03-29:
@pjroth or maybe it was some sort of an EEPROM thing that got changed along the way :)

drashna, 2018-03-29:
Possibly.  Resetting the eeprom would fix it, if that's the case.  And the ergodox keymap has the "EPRM" macro that will do this.

trunneml, 2018-04-16:
I build my own keymap from source and my ergodox-ez (Kaihl Copper Switches) started bouncing once or twice in an hour. I tried debounce 5 and it was bouncing a lot. Normaly twice in a sentence. Currently I'm trying 25 and now the keyboard feels right. I didn't had the Problem with the stock firmware, so I think it must be some of the later performance improvements.

closed 2018-10-22 by drashna