qmk/qmk_firmware #1961: ChibiOS USB Serial - Ergodox Infinity (ARM) stenography

KoviRobi, 2017-11-03:
I was trying to get my Ergodox Infinity to use the stenography extension, but I was having trouble getting it to work. The way I understand it, the problem was with it relying on LUFA, whereas the ErgoDox Infinity uses ChibiOS, which doesn't use 'virtser_send'. I was wondering if someone knows about USB serial in ChibiOS and could help me get it working?

seebs, 2017-11-22:
I think the right solution (probably?) is to implement a virtser_send for ChibiOS, so that there's one visible. But I don't know exactly how one would go about this.

drashna, 2018-03-25:
Yeah, if ChibiOS doesn't have a virtual serial port, or it properly implemented .... then "no dice". 

Maybe @fredizzimo may know what the issue is, as I believe that he's the resident ChibiOS expert.

fredizzimo, 2018-04-07:
Support for this was added in https://github.com/qmk/qmk_firmware/pull/2362

closed 2018-04-07 by fredizzimo