qmk/qmk_firmware #1987: KC_EJCT not working as X_EJCT

fruitloop, 2017-11-09:
Hi,
I recently got a V60 Type R and I since there is no Eject key I want to create a macro to lock my mac using CMD+ALT+EJECT.

I found that KC_EJCT should work as Eject key, however If I try something like:
```
SEND_STRING(SS_DOWN(X_LGUI));
SEND_STRING(SS_DOWN(X_LALT));
SEND_STRING(SS_TAP(X_EJCT));
SEND_STRING(SS_UP(X_LALT));
SEND_STRING(SS_UP(X_LGUI));
```
It gives me this error:
```
error: \x used with no following hex digits
```

Is there a way to create such a macro?

jackhumbert, 2017-11-09:
It doesn't look like [that key](https://github.com/qmk/qmk_firmware/blob/5bff9d61818c5dd692c505896aee287168af80cc/tmk_core/common/keycode.h#L423) was copied over to [the `X_` definitions](https://github.com/qmk/qmk_firmware/blob/7ad924bae5519e981c57495e481db62741aa4376/quantum/send_string_keycodes.h) - can you try `SS_TAP(B0)`? If that works, we can add the rest :)

fruitloop, 2017-11-09:
I'll try that when I'm back on my main computer tomorrow.

fruitloop, 2017-11-10:
Hi,
Now I tried the following:
```
SEND_STRING("FOO"); // just to see that I'm hitting the right key
SEND_STRING(SS_DOWN(X_LGUI));
SEND_STRING(SS_DOWN(X_LALT));
SEND_STRING(SS_TAP(B0));
SEND_STRING(SS_UP(X_LALT));
SEND_STRING(SS_UP(X_LGUI));
```

Now the build runs without errors, but when I press the key it only prints `FOO`.
The screen won't lock.

jackhumbert, 2017-11-10:
Does mapping `LALT(LGUI(KC_EJCT))` to a key work for you? That's another way of making this macro.

fruitloop, 2017-11-10:
It does!
But at work I use windows where I would need `LGUI(KC_L)` so I thought I send the mac command first set a small delay and send the windows command. That way I think I can lock both a mac and a windows machine with the same key.

fruitloop, 2017-11-15:
`LALT(LGUI(KC_MEDIA_EJECT))` works as well. `SS_TAP(X_MEDIA_EJECT)` however don't.
@jackhumbert is there anything else I can test to make this work as a macro?

drashna, 2018-03-25:
You could use `register_code` and `unregister_code` to do this, as well.  It wouldn't be as clean, but it would work .

techsock, 2018-06-19:
@fruitloop If you haven't already found a solution, I've been using this successfully to lock my Mac:

`SEND_STRING(SS_DOWN(X_LCTRL) SS_DOWN(X_LSHIFT) SS_DOWN(X_POWER) SS_UP(X_POWER) SS_UP(X_LSHIFT) SS_UP(X_LCTRL));`

fauxpark, 2018-06-20:
The definitions in `send_string_keycodes.h` appear to have exactly the same values as those in `keycode.h` - I'm not sure why the former even exists then. So you can technically use (some) `KC_` keycodes in `SEND_STRING()` as `register_code` and `unregister_code` understand media keys.

fauxpark, 2018-06-21:
Okay, I think I got it. What is actually happening to the `X_` "keycodes" is they are being [preprocessed into a hexadecimal escape sequence](https://github.com/qmk/qmk_firmware/blob/master/quantum/quantum.h#L125) eg. `\x2A`. This gets inserted as the *byte it represents* into the rest of the string. The Tap, Down and Up actions seem to be represented by the bytes `\x01`, `\x02` and `\x03`, and the keycode to action is the following byte.

The reason that `SS_TAP(B0)` doesn't work is because `B0` is already defined as the name of an I/O pin for use in the key matrix. Since it resolves to something else, the resulting string is... something other than "\xB0".

But we can avoid this conflict by just making all send_string keycode defines from A0 onwards lowercase:

```c
// keymap.c

enum custom_keycodes {
    FX_TEST = SAFE_RANGE
};

// this would go in send_string_keycodes.h
#define X_MEDIA_EJECT b0

// ...

bool process_record_user(uint16_t keycode, keyrecord_t* record) {
    if(record->event.pressed) {
        switch(keycode) {
            case FX_TEST:
                SEND_STRING(SS_DOWN(X_LGUI));
                SEND_STRING(SS_DOWN(X_LALT));
                SEND_STRING(SS_TAP(X_MEDIA_EJECT));
                SEND_STRING(SS_UP(X_LALT));
                SEND_STRING(SS_UP(X_LGUI));

                return false;
        }
    }

    return true;
}
```

closed 2018-10-22 by drashna