qmk/qmk_firmware #1991: Ergodox EZ keyboard layout debug console issue

chrisdobler, 2017-11-10:
I have tried compiling many different layouts for other keyboards, all of which correctly print debug output, except the Ergodox EZ and variants off of the master branch.

You will get 'Waiting for device:' followed by 'Listening:', and thats it. No further output is possible with this layout. I would be happy to try and fix it if anyone has any suggestions or pointers.

jackhumbert, 2017-11-10:
Does the [QMK Toolbox](https://github.com/qmk/qmk_toolbox/releases) show anything different?

hiro2016, 2017-11-10:
Is `print();` working?  
I have never managed to view debug print messages; `print`, `print_val_dec` etc are working fine though.   

rules.mk
```
CONSOLE_ENABLE=yes
DEBUG_ENABLE=yes
COMMAND_ENABLE = yes# Commands for debug and configuration
```
Magic command to toggle the debug mode on and off? `rshift + lshift+d`

jackhumbert, 2017-11-10:
Ah yeah - you should be able to the `DEBUG` keycode to do the same thing :)

chrisdobler, 2017-11-11:
Thank you guys for the prompt response!

As far as what I have tried - yes I have tried print(), printf(), debug() as well as several other variations.

I have also tried the makefile rules that you mentioned:
```
CONSOLE_ENABLE=yes
DEBUG_ENABLE=yes
COMMAND_ENABLE = yes
```
as well as
`DEBUG_ACTION = yes`

I do not currently have the i2c expander chip connected. I am wondering if perhaps that freezes the system up until it receives the signal? What do you think?

hiro2016, 2017-11-11:
>Ah yeah - you should be able to the DEBUG keycode to do the same thing :)  

Thanks I will give it a shot later.  
  
>I do not currently have the i2c expander chip connected.    

I do not know much about it but since PJRC does not mention a requirement like that for using hid_listen program and as teensy2 is a single board I feel looking into other possibilities is not that bad of an idea.  
  
+ Multiple `rules.mk` files  
Once I got caught over this, I was editing one that had lower priority.  

+  ID10T error  
Looks like you sometimes need to restart hid_listen.  
https://github.com/tmk/tmk_keyboard/issues/48  
  
+ A descriptor problem?
https://github.com/qmk/qmk_firmware/issues/726

+  bootmagic initialization?
>Note that debug/print output will not appear in hid_listen during matrix_init(), or at any time prior to Boot Magic initialization.   
https://github.com/tmk/tmk_keyboard/wiki/FAQ  
  
I'm not using the latest version of qmk, so I cannot confirm if `print` fails in the current version.  
But if all else fails, may be you can give older commits a shot; I'm using one that was downloaded in 2017 July or August and  `print` works fine on 64bit linux os with locally compiled hid_listen, 64bit compatible hid_listen is not available as executable for linux somehow, from: https://www.pjrc.com/teensy/hid_listen.html

chrisdobler, 2017-11-11:
That is an interesting note - 

> Note that debug/print output will not appear in hid_listen during matrix_init(), or at any time prior to Boot Magic initialization.
> https://github.com/tmk/tmk_keyboard/wiki/FAQ

I had tried putting debug output in the init() areas of the application as a way to debug my debug issue haha. Reason being that the i2c channel initiation does happen in the init area, so that clears up the reason it still won't output.

I suppose for the moment I'll do the programming on a different layout (standard keyboard layouts do output print(),) and order the i2c expander chip to eliminate that question later.

So regarding the i2c, this keyboard has a split design, so it sends all the keycodes from the left hand side over and i2c connection between the two keyboards. It has nothing directly to do with the communication to the computer.

chrisdobler, 2017-11-11:
Also, I am using the latest firmware and I can confirm that the print() does indeed still work.

chrisdobler, 2017-11-11:
Well what do you know? I removed all the i2c init code and I'm getting the debug output now! So problem solved. Thanks for your help guys!

closed 2017-11-11 by chrisdobler

hiro2016, 2017-11-11:
Glad to hear you have figure out the problem.  
  
>So regarding the i2c, this keyboard has a split design, so it sends all the keycodes from the left hand side over an i2c connection between the two keyboards.  
Oh, thanks. That's something I should have known.  
You seem to be well versed with hardware. I'm jealous.    
  
By the way, is debug print or `dprintf` working?  
I always assumed it does not get printed because I messed something up but secretly doubted maybe something is broken.

chrisdobler, 2017-11-11:
debug() did not work for me. I can't confirm that dprintf works, but I did look at the print library and most forms of *print*() seem to all be aliases of each other so they should work.

chrisdobler, 2017-11-11:
dprintf() does work!