qmk/qmk_firmware #2003: whitefox tap dance hangs

kwkroll32, 2017-11-12:
I have an issue where tap dance functions will hang until another key is pressed. Check out [this gist](https://gist.github.com/kwkroll32/57e72a5667cee54d32926f550ca4c8bf#file-keymap-c-L133) for an example, specifically the double tap on line 133.

Line 133 should make 1 tap of the left arrow send a KC_LEFT, and double tap send KC_HOME. If I press left arrow, I expect it to go left but it does nothing. If I tap left arrow and tap CTRL (or any other key) it works as expected and moves the cursor left (but also types whatever the other key was). The delay between left and the 2nd key does not seem to matter.

If I double tap left arrow, I get nothing just like above. If I double tap left and hit some other key, it will send a home key (and whatever the 2nd key was). 

I get the same behavior from ACTION_TAP_DANCE_FN as I do with ACTION_TAP_DANCE_DOUBLE (line 132). It seems like the tap dance functions are stuck waiting for something, and pressing another key allows it to move along. I have the same function working on a native QMK M65-A.

bwright86, 2017-11-15:
I have the same issue with a handwired Planck keyboard, running a teensy 3.2. It appears that after performing the dance, a second key press is needed to register the results of the dance.

Not sure if that is the way this feature was built, but it is the results I am observing.

For clarity, I am using the ACTION_TAP_DANCE_DOUBLE and ACTION_TAP_DANCE_FN_ADVANCED methods for Tap Dance.

kwkroll32, 2017-11-18:
Does the teensy also use chibios? If so, that is a constant between our boards, and something different compared to my other board where tapdance works as expected. 

I could try flashing a tapdance function to my QMK ergodox infinity if we need more testing - pretty sure that uses chibios too.

bwright86, 2017-11-18:
Yes, it is a Teensy 3.2, so it uses ChibiOS. You could have a point, since this is my first qmk keyboard, I do not know what to expect with these features, other than what I have observed.

I have an infinity ergodox too. One day I will have to try porting my configuration over to qmk, and try some new features out (Currently it is my daily driver.)

The infinity ergodox uses the exact same chip as the Teensy 3.2, came across that tidbit when setting up my new Planck keyboard.

drashna, 2018-03-25:
Did you define the TAPPING_TERM?

kwkroll32, 2018-03-27:
I did not have TAPPING_TERM defined. I added this config.h to my whitefox keymap directory and now it works. Thanks for the tip. 

`#include "../../config.h"`
`#undef TAPPING_TERM`
`#define TAPPING_TERM 200`

closed 2018-03-27 by kwkroll32

drashna, 2018-03-27:
Glad to hear it!