qmk/qmk_firmware #2005: `dfu` target doesn't have any effect / output

20015jjw, 2017-11-13:
Hey, I'm getting this weird behavior when I try to flash the qmk onto my tmk whitefox. 

```
willj-mbp:qmk_firmware willj$ sudo make whitefox:jetpacktuxedo:dfu
QMK Firmware 0.5.119
Making whitefox with keymap jetpacktuxedo and target dfu

willj-mbp:qmk_firmware willj$
```

I'm sure I've pulled all the lastest change, rerun all the `homebrew` commands yet I'm not getting anything after this command. The above command works fine without `:dfu` but it doesn't flash the firmware onto the keyboard. I think it worked fine ~2 month ago, but I'm not sure if it had `:dfu` thingy added back then. 

I've also tried the `qmk_toolbox` and it also doesn't see the keyboard when it's in the DFU mode. 

```
*** QMK Toolbox (http://qmk.fm/toolbox)
    Supporting following bootloaders:
     - DFU (Atmel, LUFA) via dfu-programmer (http://dfu-programmer.github.io/)
     - Caterina (Arduino, Pro Micro) via avrdude (http://nongnu.org/avrdude/)
     - Halfkay (Teensy, Ergodox EZ) via teensy_loader_cli (https://pjrc.com/teensy/loader_cli.html)
     - STM32 (ARM) via dfu-util (http://dfu-util.sourceforge.net/)
     - Kiibohd (ARM) via dfu-util (http://dfu-util.sourceforge.net/)
*** TMK  - WhiteFox  connected -- 0xFEED:0xF0F
  ! There are no devices available
*** TMK  - WhiteFox  disconnected -- 0xFEED:0xF0F
  ! There are no devices available
*** TMK  - WhiteFox  connected -- 0xFEED:0xF0F
  > USB configured.
  > bootmagic scan: ... done.
    Keyboard start.
  ! There are no devices available
  ! There are no devices available
  ! There are no devices available
  ! There are no devices available
*** TMK  - WhiteFox  disconnected -- 0xFEED:0xF0F
  ! There are no devices available
  ! There are no devices available
  ! There are no devices available
  ! There are no devices available
*** Downloading the file: http://qmk.fm/compiled/whitefox_default.hex
    File saved to: /Users/willj/Downloads/whitefox_default_88DEED75.hex
  ! There are no devices available
```

Any help is appreciated. Thanks in advance.

jackhumbert, 2017-11-13:
I believe the whitefox needs `:dfu-util` to be programmed - does that do anything? I think it needs to be put into the bootloader before you'll be able to program it though.

kwkroll32, 2017-11-14:
I haven't tried the `:dfu-util` way, but I flashed QMK onto my R1 whitefox by doing 

`sudo make whitefox:jetpacktuxedo:bin`

this produces a bin file for your desired keymap, which I then flash onto the board by hitting the reset button on the bottom (you should probably see an orange LED light up through the reset hole when the board is ready to flash) and executing 

`sudo dfu-util -D whitefox.bin`

If you need more info about the 2nd command, the [kiibohd documentation](https://github.com/kiibohd/controller/wiki/Loading-DFU-Firmware) might help.

20015jjw, 2017-11-14:
@jackhumbert Thanks. I tried the command without the keyboard attached and it does show a familiar interface that ends with `No DFU capable USB device available`. So I guess it works now. Thanks. BTW, is this anywhere in the doc? I didn't find it and I assume it might give others a hard time if it's not there. 

@kwkroll32 Thanks. I was going to do it manually like you did. Now `:dfu-util` works I think.

closed 2017-11-14 by 20015jjw

kissu, 2018-12-31:
Update for @kwkroll32's dead link above available [here](https://kiibohd.github.io/wiki/#/Flashing?id=flashing-linux). :ok: