qmk/qmk_firmware #2011: Tapping `LT(2, KC_SPACE)` before holding results in two spaces...

Camsbury, 2017-11-15:
The firmware seems to give up on the possibility of the held key being held if the most recent press was a tap. I use my space key as a layer switch currently, and it would be good to be able to use that quickly after typing space, but I'm running into the issue that it wants to just repeat spaces after the first tap.

Any tips? As I said above the key mapping is `LT(2, KC_SPACE)` for my ergodox_ez keyboard.

Camsbury, 2017-11-15:
It seems someone also posted about this on [reddit](https://www.reddit.com/r/olkb/comments/77hy9e/qmk_mod_tap_fast_typing_issues/).

hiro2016, 2017-11-15:
Someone said changing tapping term helps:
```
#define IGNORE_MOD_TAP_INTERRUPT 
#define TAPPING_TERM 150
```
https://github.com/qmk/qmk_firmware/issues/1819
   
If the above does not help, use a macro/custom keycode till the author fixes it.  
LT probably uses a counter in matrix scan to differentiate between tap and hold and I got a haunch that counter is the tap count, since disabling tap dance also disables  LT.     
  
Ya, LT is great but it can be better.
Though even understanding it does not seem like an easy job, code jumps all over and hard to track.

Camsbury, 2017-11-15:
Yeah I've adjusted `TAPPING_TERM`, but that doesn't affect this issue at all. Something about double tapping a "mod tap" key seems to discard the mod functionality completely. I read somewhere people might combo this with tapdance, so that one tap is space/LT, two is space followed by LT, and three is just space, allowing for the repeat. Gotta figure out how to implement that now...

hiro2016, 2017-11-15:
Hmmm,  even with that set up using LT after typing space for 4 times...might be a little hard.  
As I have suggested to the other guy, a custom keycode should be a better solution.  
```
enum custom_keycodes {
  LT_ONE_OR_ENTER = SAFE_RANGE
};
uint16_t custom_lt_timer;
keypos_t prv_key_pressed;
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
#define IS_KEYPOS_SAME(keyone,keytwo)  ((keyone.col==keytwo.col)&&(keyone.row==keytwo.row))
#define ANOTHER_KEY_PRESSED (!IS_KEYPOS_SAME(prv_key_pressed, record->event.key))
       if(record->event.pressed) prv_key_pressed = record->event.key;
	if(keycode == LT_ONE_OR_ENTER) {
		if(record->event.pressed) {
			custom_lt_timer = timer_read();
			layer_on(1);
		}else{
			layer_off(1);
			if (timer_elapsed(custom_lt_timer)<200 && (!ANOTHER_KEY_PRESSED)) {
                            register_code(KC_ENT);  
                            unregister_code(KC_ENT);
			}
		}
	return true;
}
```

typester, 2018-01-15:
I had the same issue and couldn't solve it by any configuration.
I gave up and wrote following dirty patch to solve this problem.
https://gist.github.com/typester/38cc078b38e71ea1c09da7a58e656de0

This patch does force reset TAPPING_TIMER when the tap key is released so I can't use tap dance anymore. It's no problem for me, I don't use it.

Camsbury, 2018-01-17:
It looks like a fairly simple patch, at the least. I'll try it out and let you know.

drashna, 2018-05-23:
Oh, `TAPPING_FORCE_HOLD` will fix this. But it breaks tap toggles.

closed 2018-10-22 by drashna