qmk/qmk_firmware #2026: Ergodox Infinity - backlight only works on one hand

plattfot, 2017-11-19:
Hi,
I'm having some issues with getting the backlight to work properly. Had no issues with the backlight when I first got my ergodox infinity in september and flashed qmk on it. It was working fine for both hands. 

But after syncing my [fork](https://github.com/plattfot/qmk_firmware/tree/master/layouts/community/ergodox/plattfot) with your master branch, made some tweaks to my layout and updated my keyboard the backlight started to have issues. When I use my latest keymap only the right hand works, if I go back to my old keymap only the left works... 

If I flash the right hand to my left hand the backlight works. I even got it working for both hands this morning but I needed to make some more changes to my layout and now I'm back to only the right is working.
Steps I'm doing when building are:

Unplug the keyboard, insert only the left hand, press the flash button
`sudo make ergodox_infinity:plattfot:dfu-util`
Unplug the left hand, insert the right, press the flash button
`sudo make ergodox_infinity:plattfot:dfu-util MASTER=right`

I've tested cloning the latest qmk_firmware, copy my layout over ([plattfot.v4](https://github.com/plattfot/qmk_firmware/tree/plattfot.v4/layouts/community/ergodox/plattfot)) and build. Only right hand's backlight is working.
Cloned a new copy of my fork, checkout an older version of my firmware ([plattfot.v1](https://github.com/plattfot/qmk_firmware/tree/plattfot.v1/layouts/community/ergodox/plattfot)) and build. Only the left hand's backlight is working.

Everything else works perfectly, no key issues, lcd works etc. 

I've tried swapping the cables, i.e. the left cable go to the right and vice versa in case one cable was bad. 

Any ideas what it can be? First I thought it might be hardware issues (might still be), but that doesn't explain that I can get it to work if I flash the right hand's firmware to my left hand. Looks to be similar to #1415 but that should be fixed.

ieure, 2017-12-09:
I'm also experiencing this issue.  It was working on both hands when I initially switched to QMK, but now only the left hand lights.

plattfot, 2018-01-06:
Did another merge with upstream today, [81ae0fb1](https://github.com/qmk/qmk_firmware/commit/81ae0fb10efb10414a7578f374e5fbbff58984d4). Backlight now works on both hands for me.

tafryn, 2018-01-10:
I am experiencing this issue, too. All backlight commands only affect the left hand regardless of which side is the master.

drashna, 2018-03-25:
I believe that this was fixed in #1776 

Try updating your repo and see if the issue persists

drashna, 2018-10-22:
Sounds like this was fixed, or abandoned.

If not, please reopen.

closed 2018-10-22 by drashna