qmk/qmk_firmware #2033: Steno protocol not creating serial port

KeyboardFire, 2017-11-19:
I have followed the instructions on [the Stenography page](https://docs.qmk.fm/feature_stenography.html) of the QMK documentation by configuring both my QMK layout and Plover for steno over virtual serial port. However, I can't get any output with either TX Bolt or Gemini. Nothing happens when I press one of the steno keys or try to stroke a chord.

After some investigation, it seems that no virtual serial port is being created, based on the output of `dmesg`.

[Here are my `keymap.c` and `rules.mk`](https://gist.github.com/KeyboardFire/02f72d249f5cae13030cf5af23af24b6); I've also tried the `steno` layout from the QMK repository with the same (non-)results.

The `dmesg` output when I unplug and reconnect my Planck is as follows:

```
[400494.637423] usb 1-1: USB disconnect, device number 43
[400496.242116] usb 1-1: new full-speed USB device number 44 using xhci_hcd
[400496.376597] input: OLKB Planck as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:FEED:6060.0049/input/input93
[400496.432780] hid-generic 0003:FEED:6060.0049: input,hidraw2: USB HID v1.11 Keyboard [OLKB Planck] on usb-0000:00:14.0-1/input0
[400496.434214] input: OLKB Planck as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:FEED:6060.004A/input/input94
[400496.489155] hid-generic 0003:FEED:6060.004A: input,hidraw3: USB HID v1.11 Device [OLKB Planck] on usb-0000:00:14.0-1/input1
[400496.490715] input: OLKB Planck as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:FEED:6060.004B/input/input95
[400496.546131] hid-generic 0003:FEED:6060.004B: input,hidraw4: USB HID v1.11 Keyboard [OLKB Planck] on usb-0000:00:14.0-1/input2
```

`dmesg | grep tty` yields only the following:

```
[    0.000000] console [tty0] enabled
[    1.561350] 0000:00:16.3: ttyS0 at I/O 0x30b0 (irq = 17, base_baud = 115200) is a 16550A
```

And here's the output of `xinput --list`, in case that helps:

```
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Synaptics tm2964-001                    	id=17	[slave  pointer  (2)]
⎜   ↳ TPPS/2 IBM TrackPoint                   	id=18	[slave  pointer  (2)]
⎜   ↳ Logitech USB Receiver                   	id=12	[slave  pointer  (2)]
⎜   ↳ Logitech USB Receiver                   	id=20	[slave  pointer  (2)]
⎜   ↳ OLKB Planck                             	id=11	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=8	[slave  keyboard (3)]
    ↳ Integrated Camera: Integrated C         	id=14	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=15	[slave  keyboard (3)]
    ↳ ThinkPad Extra Buttons                  	id=16	[slave  keyboard (3)]
    ↳ Logitech USB Receiver                   	id=13	[slave  keyboard (3)]
    ↳ OLKB Planck                             	id=9	[slave  keyboard (3)]
    ↳ OLKB Planck                             	id=10	[slave  keyboard (3)]
    ↳ OLKB Planck                             	id=19	[slave  keyboard (3)]
```

This is what my Plover config looks like (if I try "Scan," the only result is `/dev/ttyS0`):

[image]

I'm using a Planck on Arch Linux with Plover 4.0.0.dev3 and the latest version of the QMK repository.

Thanks!

seebs, 2017-11-22:
I'm not seeing the rules.mk (it's just a link to keymap.c, and trimming the /keymap.c shows a list of things not including a rules.mk).

My rules.mk for my keyboard using this just has `STENO_ENABLE = yes`, and that seems to work.

```
[1228506.880367] input: ErgoDox EZ ErgoDox EZ as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7.1/1-7.1:1.0/0003:FEED:1307.036D/input/input590
[1228506.936069] hid-generic 0003:FEED:1307.036D: input,hidraw0: USB HID v1.11 Keyboard [ErgoDox EZ ErgoDox EZ] on usb-0000:00:14.0-7.1/input0
[1228506.937941] input: ErgoDox EZ ErgoDox EZ as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7.1/1-7.1:1.1/0003:FEED:1307.036E/input/input591
[1228506.991697] hid-generic 0003:FEED:1307.036E: input,hidraw1: USB HID v1.11 Device [ErgoDox EZ ErgoDox EZ] on usb-0000:00:14.0-7.1/input1
[1228506.993273] hid-generic 0003:FEED:1307.036F: hiddev0,hidraw2: USB HID v1.11 Device [ErgoDox EZ ErgoDox EZ] on usb-0000:00:14.0-7.1/input2
[1228506.993491] cdc_acm 1-7.1:1.3: ttyACM0: USB ACM device
```

KeyboardFire, 2017-11-22:
@seebs The `rules.mk` is on the same page as the keymap (scroll down to the bottom). It contains `STENO_ENABLE = yes`, but that doesn't seem to be working.

seebs, 2017-11-22:
Hmm. Okay, experiment time: Try turning off NKRO_ENABLE, and possibly CONSOLE_ENABLE. There's code to check for running out of endpoints, but then I noticed:

`OPT_DEFS += -DINTERRUPT_CONTROL_ENDPOINT`

And I don't know the endpoint stuff well, but it occurred to me that I could easily imagine the system being smart enough to detect features with too many endpoints, but not smart enough to detect that this -D is also adding an endpoint...

seebs, 2017-11-22:
I checked more, that *probably* isn't it, because there's other keyboards with the same thing that aren't doing this. So I'm not sure. I'm still a little suspicious about the endpoint thing.

Do other USB serial things work on this system? Is it possible that you have a kernel that omits the relevant support? I know arch users (like me...) have been known to optimize a kernel by dropping hardware they didn't need...

KeyboardFire, 2017-11-23:
@seebs Ah, good call, it was indeed a kernel thing. I had forgotten to reboot after upgrading my kernel, so the required modules probably weren't getting loaded. Sorry about that, and thanks!

closed 2017-11-23 by KeyboardFire

co-dh, 2018-01-28:
For me, I need to run plover with sudo.  I guess the virtual serial is not accessible from normal user.

seebs, 2018-03-13:
Serial port accessibility is a configuration choice you'd make elsewhere; usually there's a group that gets access to the port, and you'd need to add your user account to it. You should not need root to access a serial port in normal use.