qmk/qmk_firmware #2036: No keycode to disconnect bluetooth connection from current device.

harshitgoel96, 2017-11-20:
I am currently working on a magicforce 68 mod to make it bluetooth capable. I am useing Bluefruit BLE 32u4. Everything is working fine, and I can easily use keyboard in bluetooth mode. The issue i have is when using multiple devices like phone or laptop, i need to disable the connection from the Device, i.e. laptop or mobile, is it possible to force disconnection for the keyboard itself ? i could not find a keycode or something that would make it possible.

drashna, 2018-03-25:
Doesn't look like there is a way to do this currently. 
Switching between USB and BT mode would probably be a reasonable hack.

harshitgoel96, 2018-03-26:
that is something i am currently using. thanks. i will close this request. as i half expected a resolution on this.

drashna, 2018-03-26:
If you don't mind, I'm going to reopen it, and tag it.  Adding an actual disconnect or switch command would be a really good idea.

Miodec, 2018-06-05:
Is there any progress with this feature? Is it even possible? Also @drashna im not sure what you mean by "switching between USB and BT is a reasonable hack" as this still leaves the board connected to the device and just stops sending keycodes. It doesnt actually disconnect, which still forces the user to dig through the settings of that device and forgetting the keyboard, before it can be connected to a new device.

jackhumbert, 2020-03-17:
I think that solution is probably sufficient for now, until we get broader support for BT modules - you may want to track #2054.

closed 2020-03-17 by jackhumbert

ts52, 2022-01-04:
I'm confused. The OUT_USB, OUT_BT 'solution' doesn't work for me on an Adafruit Bluefruit 32u4. And 2504 seems to only be asking which boards are supported. But having a keycode that specifically puts the bluetooth in pairing mode would be extremely useful.