qmk/qmk_firmware #2037: Reversed macro layer

QuadDepo, 2017-11-21:
Hey 👋 

I had this idea of having a sort of Reversed macro layer, sorry a what?
yea maybe we should change the name but let me explain the concept.

We all the know in the action movies when you wanna launch a rocket missile you need two keys to activate it. 

[image]

I was thinking about having the same concept to activate a layer.

So if I press my **left shift key** and my **right shift key** at the same time or holding them both down, I go to **layer 8** for example.

I'm not 100% sure this doesn't exist yet, but I couldn't find anything about it in the documentation.

Let me know what you guys think about it.

_Happy Hacking :-) _

Ryan

drashna, 2017-11-21:
The Command Keys can do this (usually LSHIFT+RSHIFT+key). 0-9 to get to a specific layer.  You just need to make sure there is a way to get back.

Otherwise, you could do this with the `process_record_user` function.  Just create a variable, increment it up when the two are pressed, check the value and switch to the layer if it's "correct".

QuadDepo, 2017-11-21:
@drashna Yea, I wasn't sure if there was already something like it. Well, my idea was also that it was gonna be a MO layer, so when you release the keys you go back to the layer you came from.

Can you explain the "try_layer" a little more, I couldn't find anything about on the wiki.

drashna, 2017-11-21:
Then the tri layer thing is probably best

https://github.com/qmk/qmk_firmware/blob/master/keyboards/planck/keymaps/default/keymap.c#L201

RAISE/LOWER goes to their respective layer.  The `update_tri_layer` function then switches to the `adjust` layer if both the RAISE and LOWER layers are active (aka pressed+held). 

If you want it "stealthy" (as in, nothing changes until both are pressed), then you'd want to make sure the two layers are full of "KC_TRANSPARENT" keycodes.

QuadDepo, 2018-03-25:
I did :)

closed 2018-03-25 by QuadDepo

drashna, 2018-03-25:
Awesome! 
Did you make a PR yet, so we can all see your glorious code? :)

QuadDepo, 2018-03-25:
Currently have a own repo here :) https://github.com/QuadDepo/niu_mini

drashna, 2018-03-25:
Awesome! 
Thanks for the link. 

If you could, I'd recommend setting it up to make a pull request, to get it added to the official repo.  
I can do this for you, if you'd like.  But I think it would be awesome for others to be able to see and use it. :)