qmk/qmk_firmware #2053: Already pressed key gets released when switching layers

feidaman, 2017-11-25:
Full disclosure: I'm a newbie using an Ergodox EZ here, haven't actually built my own firmware (yet). I have a key configured as **Ins** on **Layer 0**. I also have another key mapped to MO(1). If I press and hold Ins, as soon as I press the layer toggle key Ins up is sent, but the key is still being physically held down. Wouldn't it be useful if QMK could remember the layer where key-down occurred, and preserve the context through to key-up?

notg10, 2018-07-13:
Can we revisit this issue? Heres a perfect use case:

Holding W to walk in a game and using fn+1 for f1 and you stop moving until you re-hit w.

this is less then ideal and to my knowledge no other firmware has this limitation.

thanks,

feidaman, 2018-07-13:
Is this really because of limited RAM? From a users  respective, the way this works ATM is wrong---If the key is physically down, key up should not be sent automatically for me.

closed 2018-12-28 by mechmerlin

drashna, 2018-12-28:
This has been fixed now.  Update your repo, and it should be the default behavior.

You can revert by defining `STRICT_LAYER_RELEASE`.

jackokring, 2021-11-20:
Hi. Nice to option but old builds could do with the old behaviour. As games and industrial process control systems have different ideas about safety of unexpected stuck keys. Any plans for a source build date #define?

PR #3905 (2018-12-28): Keep pressed keys on layer state change (fixes #2053, #2279)
This allows pressed keys to stay pressed when a layer change occurs.

Say you're playing a game and you're holding W to run forward. You need to access a menu or skill that's bound to F4. You have that set on a momentary layer, so you try pressing Fn+4 — but as soon as you press Fn, QMK resets the W key and you stop moving forward, even though you're still holding the key.
For a visual explanation of the issue, see [this video](https://youtu.be/O4AKOX7F5SA?t=4h30m34s).

This fixes the problem by not resetting held keys when the layer [... cut]
commit 93b004c9 (2018-12-28): Keep pressed keys on layer state change (fixes #2053, #2279) (#3905)

* Keep pressed keys on layer state change

* Add doc comment for clear_keyboard_but_mods_and_keys

* Keep pressed keys only if PREVENT_STUCK_MODIFIERS is on

* Check STRICT_LAYER_RELEASE instead of PREVENT_STUCK_MODIFIERS