qmk/qmk_firmware #2054: Which Feather Bluetooth boards are compatible.

SleepyBrett, 2017-11-25:
Adafruit sells at least three bluetooth feather boards and I'm having a hard time figuring out which are compatible with the firmware (with bluetooth support of course).

Here is their breakdown: https://learn.adafruit.com/adafruit-feather/bluetooth-feathers

I will summarize as best I can:

- Feather nRF52 - nRF52832 MCU (ARM Cortext M4F), LE only (single SOC)
- Feather Bluefruit M0 - ARM M0, nRF51, LE only
- Feather 32u4 Bluefruit LE - ATmega32u4, nRF51, LE only
- ESP32 Feather - Bleeding edge may or may not have BT support  yet ( did not as of may )

Maybe there is a page of supported boards somewhere in the documentation, but I couldn't find it.

shelaf, 2017-12-14:
Feather 32u4 Bluefruit LE only supproted.
Please see below issue about M0.
https://github.com/qmk/qmk_firmware/issues/912

shelaf, 2017-12-14:
Using a Bluefruit Ez-Key via qmk supported board like an atmega32u4, probably work well. But, I've never tried.
https://www.adafruit.com/product/1535

drashna, 2018-03-25:
Would be nice to add a supported and tested list to the docs.

un4tuner, 2018-11-23:
Any plans to support Adafruit nrf52?

yiancar, 2018-12-12:
The adafruit nrf52 uses the Nordic MCU nRF52832 as described above. For this to happen, chibios needs to get configured to use the propitiatory Nordic Bluetooth SDK.

Furthermore the nRF5 cannot be a usb client. It can only be used to program the nRF52832.

If you want to go to that route, I would suggest finding a board with an nRF52840 in it. Getting the SDK to work with chibios is the main priority if anyone wants to give it a go. If so Nordic themselves sell a really really good devkit for the nRF52840.

joric, 2019-02-25:
Check this out https://github.com/sekigon-gonnoc/BLE-Micro-Pro
And maybe my https://github.com/joric/nrfmicro/
And especially https://github.com/sekigon-gonnoc/qmk_firmware/tree/nrf52
TL;DR: sekigon keyboard guy added nrf51/nrf52 support as a new QMK platform for bluetooth split keyboards.
There's no Atmega32u4 involved, only nRF5x (including 52840 with full USB support) chipsets.
Managed to compile it using Nordik SDK 12, it compiles just fine, going to add my keyboards.
I wish it would be merged into the main branch someday.

yiancar, 2019-02-25:
@joric do you have a 52840 devboard?

joric, 2019-02-25:
Nope, I'm waiting for two E73-2G4M08S1C nRF52840 modules in the mail, I don't have a devboard, going to debug everything on my nrfmicro (boards should arrive in a few weeks). I have a couple of unassembled EByte_E73_Batt_ErgoTravel boards (nRF52832-based, no USB support) from here https://github.com/jpconstantineau/NRF52-Board probably will try them first.

yiancar, 2019-02-25:
Yes I am quite more interested in the nRF52840 to integrate the nordic sdk usb lib

ELLIOTTCABLE, 2019-04-17:
I'm so happy to see that the nRF52840 is getting some play with QMK! I never ended up having time to work on this, but it was important to me — I'm gonna have to flash my monster of a prerelease devboard, and dick around with it a little.

Such good news; great work, @sekigon-gonnoc!

joric, 2019-04-17:
Check out my nrfmicro and Jian BLE firmware, by the way https://github.com/joric/nrfmicro/wiki

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc