qmk/qmk_firmware #2059: ACTION_LAYER_TAP_KEY is toggling layer, not momentarily activating it

plantarum, 2017-11-25:
Hi,

I'm using ACTION_LAYER_TAP_KEY on my space key in the atreus map, with the intention of momentarily providing access to my third layer. FN0 is defined as:

const uint16_t PROGMEM fn_actions[] = {
    [0] = ACTION_LAYER_TAP_KEY(_LW, KC_SPC), // to layer 2 while held, space when tapped
};

It works as expected, except that the layer is toggled on, not momentarily accessed. That is, after releasing the space key I'm still on layer 2 until I press a key bound to TO(0).

I'm also using OSM and OSL, if that has any effect. My keyboard.c is here:

https://github.com/plantarum/qmk_firmware/blob/master/keyboards/atreus/keymaps/tws2/keymap.c

Am I missing something in my config to get this working normally?

Thanks!

drashna, 2018-03-25:
That's very odd.   Are you still having this issue?

drashna, 2018-03-25:
Ah, yeah.  Basically, it was getting stuck because it wasn't registering the upstroke, ever. 

Glad to hear you got it resolved!